wavelog/wavelog #1144: Advanced Logbook: "Transmit Power" Column missing, also missing in "Edit"

xyz667, 2024-11-01:
In the Advanced Logbook there is no column for "Transmit Power".
Also, "Transmit Power" ist missing from "Edit", so it is impossible to do mass changes for this QSO field.

Version Information

OS: DietPi 9.7.1 (Bookworm)
PHP: 8.2.24
MySQL/MariaDB: 10.11.6-MariaDB-0+deb12u1
Wavelog: 1.8.8

Steps to Reproduce
Open Advanced Logbook (no column for Transmit Power).
Try to change Transmit power via the Edit button (Transmit Power missing from dropdown selection options).

Expected Behavior
Transmit Power showing in Advanced Logbook and possible to change via edit button.
This is necessary to change the differing transmit power for digimode QSOs, which are logged with the default power via the API (the digimode program is not sending that info).

Desktop Information:

OS: macOS 15.0.1
Browser: Safari

xyz667, 2024-11-02:
Please not forget to add any new columns to "Options for the Advanced Logbook" as well. ;-)

AndreasK79, 2024-11-02:
It's all in #1146.

AndreasK79, 2024-11-04:
@xyz667 the code is in dev, if you run that branch. If not, wait for release :)

closed 2024-11-12 by github-actions[bot]