wavelog/wavelog #1174: Create Logbook Filter for "Where Grid is empty"

HochdruckHummer, 2024-11-08:
It's currently not possible to search for fields that are empty. (Or I didn't find out how to do that).

Please provide a filter to show only QSOs with missing grid. That way I would be able to batch select all search results for Callbook update.

dg9vh, 2024-11-08:
Hi,
[image]
Using advanced search would bring you to your target.

HochdruckHummer, 2024-11-08:
I'm sorry... How do I get to advanced search?

HochdruckHummer, 2024-11-08:
This small moment of joy "Yay, this function is already given, I'm so happy!" to "Naaah, doesn't work".... :)

dg9vh, 2024-11-08:
but for locating them, you should use following:
[image]
then you know them :-) but how to actualize them in batch from qrz.com is open - so this could be a real feature request. @HB9HIL @phl0

dg9vh, 2024-11-08:
We just need a mass editing function (like in advanced log) here in advanced search based on the search result.

HB9HIL, 2024-11-10:
It's probably easier to add this "is empty" filter in Logbook Advanced then add Batch Edit/Update From Callbook in the Advanced Search.

dg9vh, 2024-11-10:
Well, yes... correct :-) but in advanced log you do not get all "in one query"... only up to 5000

AndreasK79, 2024-11-10:
As an empty filter is a good suggestion, I will add that. Probably with a dropdown, so you can choose which column.

AndreasK79, 2024-11-19:
Added PR here #1220. With this change, you can search for empty values where the filter is a textbox.

AndreasK79, 2024-11-19:
Merged into dev for those wanting to test.

closed 2024-11-28 by github-actions[bot]