wavelog/wavelog #1176: Make " n times worked before " table configurable.

pa8s, 2024-11-08:
**What Feature is Missing?**
The " n times worked before " table, as seen in the Post or Live QSO page only has a QSL column. I would like these columns to be configurable in a similar way as the 'Logbook fields' in my account 'Wavelog Preferences'.

**Why is This Feature Important?**
With this I can replace or add a column in the " n times worked before " table so when entering a QSO I could see if I have already a LoTW QSL from this call. Similar to the current QSL column (which only shows QSL cards if I'm correct).

**Is This Feature Personal or Beneficial to Others as Well?**
I think this can be beneficial to others.

phl0, 2024-11-29:
Maybe I am not understanding correctly. There is a user option called "Previous QSL Type" which allows you to select which QSL type (e.g. paper, LoTW, etc.) is shown in the worked before section.

[image]

Is this what you are looking for? If not please be a little more specific.

pa8s, 2024-12-01:
Ah, that is a configuration item I've totally overlooked, I've set it to LoTW and now it is exactly what I want. Well, than this enhancement request can be closed.

closed 2024-12-01 by pa8s