wavelog/wavelog #1189: keep clean the Location + Gridsquare Fields during /p, /m, /mm activities

dg2ron, 2024-11-12:
Is it possible to keep the Location and Gridsquare fields free during /p, /m or /mm? 
You are never at the specified location when you are /p, /m or /mm

I have noticed that these two fields are often filled in. Whether this data comes from QRZ or from past QSOs remains to be determined. In my opinion, these fields should be left blank and filled in optionally.

HB9HIL, 2024-11-12:
Hello @dg2ron 

Are you talking about the QSO logging or the station setup?

Can you precise your question?

dg2ron, 2024-11-12:
QSO logging

[image]
...
[image]

I think he is pulling the "wrong location" from a past QSO.

HB9HIL, 2024-11-12:
Please tell me:

- Which Version of Wavelog are you running? This issue was fixed in #452  and released in Version `1.7`
- Which callbook is configured? `qrz` or `hamqth`

int2001, 2024-11-12:
@HB9HIL he uses latest dev. qrz is configured.

HB9HIL, 2024-11-12:
@dg2ron @int2001 

I was able to reproduce this issue. 

@dg2ron: It shows you the value `/P Storkow` because you once logged exactly this callsign (incl. `/P`) with this value in the past. 

@int2001 We probably should omit that. I will check that out

int2001, 2024-11-12:
if we omit that, than only when logging things with dynamic location (like `/P` or `/MM` or `/AM`).
Otherwise the whole caching-logic for QRZ will break.

HB9HIL, 2024-11-12:
Yep thats the idea

HB9HIL, 2024-11-12:
@dg2ron PR is merged. Please confirm the issue fixed after update

@int2001 your turn

dg2ron, 2024-11-12:
TNX! 
cfm! All okay now. :)

closed 2024-11-12 by HB9HIL