wavelog/wavelog #1197: Cabrillo - export needs to be fixed to be compatible

205er, 2024-11-12:
Today I joined the DARC FT4 contest and tried to export the QSO to a Cabrillo file.
Unfortunately, the created file did not comply with the requirements of the DARC contest hub (https://dxhf2.darc.de/~ft4log/upload.cgi?form=referat&lang=en)
The following issues appeared and needed to be fixed manually:
1. _Mode_ must not be "MFSK" but "DG" when logged QSO was in FT4. As far as I know MFSK is never allowed.
2. QSO must contain _exch_ with value of _locator(S) & RST (S)/locator(R) & RST (R)_

I attached the original file and my adjusted file
[adjusted.cbr.txt](https://github.com/user-attachments/files/17722763/adjusted.cbr.txt)
[original.cbr.txt](https://github.com/user-attachments/files/17722764/original.cbr.txt)

Any chance to get this changed or at least make the export customizable?

My used Wavelog version: 1.8.7 

Thanks

HB9HIL, 2024-11-12:
Hello @205er 

You're right. Related to the official Cabrillo Documentation it should be `DG` and not `MFSK`

https://wwrof.org/cabrillo/cabrillo-qso-data/

205er, 2024-11-13:
*wow*, that was fast!
Thanks for your great support!