wavelog/wavelog #1199: ADIF export for Advanced Logbook filtered view

xyz667, 2024-11-13:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
In the Advanced Logbook it is possible to filter the display, there is no way to export all matching QSO to ADIF, however.
Thus, I recommend to add a (filtered) export functionality to the Advanced Logbook.
Alternatively, add a filter to the ADIF export menu option.

**Why is This Feature Important?**
Currently it is only possible to Export *ALL* QSO in the database to ADIF.
Sometimes I'd like to be able to export only a subset of certain QSO.

**Is This Feature Personal or Beneficial to Others as Well?**
I would expect others could use that feature too, e.g. for importing only certain QSO to DARC DCL or something alike, or transferring only digimode QSO to the WSJT-X log or whatever.

AndreasK79, 2024-11-13:
That is exactly how the ADIF export does work in the advanced logbook. Set up filters and export. The code will then export all QSOs with that filter. If you mark QSOs, only the marked ones will be exported.

xyz667, 2024-11-13:
@AndreasK79 oops I have totally missed that button.
Another indicator there's something about #1201  ;-)

closed 2024-11-13 by xyz667

AndreasK79, 2024-11-13:
Good that you found it😊