wavelog/wavelog #1206: LoTW upload fail when uploading 2 identical QSOs from 2 differents stations

F4JSU, 2024-11-14:
**Describe the bug**
I mainly activate POTAs. Occasionally, a zone may have several POTAs (e.g. 2, 3).
As it's not possible to put several POTAs in the same station, so I create two stations.
The bug is on LoTW synchro.
Only one QSO is sended _(I think is the right choice)_, but if i check the QSO on Wavelog it says "sended" for both stations. It isn't the case.

**Version Information**
- OS: Debian 12
- PHP: 8.3
- MySQL/MariaDB: 10.11.6-MariaDB-0+deb12u1
- Wavelog: 1.8.2

**Steps to Reproduce**
1. Create station FR-2933
2. Copy station FR-2933 and create station FR-2934
3. Load station FR-2933 with adif file
4. Load station FR-2944 with adif file
5. Synchronize with LoTW wavelog/index.php/lotw
6. Run "Manual Sync"
7. Wait for other amateurs to send their QSOs.
8. Search for a callsign in the FR-2933.
9. Watch LoTW status

**Expected behavior
Depends on :
1. Send both QSOs to LoTW, cons: other amateur will send only one QSO.
2. Allow multiple POTA references for one station, cons: you may have to change part of Wavelog.
3. Don't change too much, try to link the QSO from station A to station B, kind of symbolic link, cons: it's a quick fix.

**Screenshots**
View in Wavelog
[image]
LoTW view
[image]

**Additional Context**
You may have the same problem for other OTA activities.

phl0, 2024-11-14:
LoTW does not handle POTA references for confirmations. Wavelog allows to configure multiple POTA references per station location. You should add them comma separated.

F4JSU, 2024-11-14:
Ok, thanks, so my fault, 73!

closed 2024-11-14 by F4JSU