wavelog/wavelog #1213: Mode USB not shown in filters in advanced logbook view

sebket, 2024-11-17:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
When filtering QSO entries after mode, the mode USB is not displayed in the dropdown menu.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Ubuntu Server 24.04.1 LTS 
- PHP:              8.3.6 
- MariaDB:     10.11.8
- Wavelog:      1.8.8

**Steps to Reproduce**
1. go to Logbook > Advanced
2. Location: select all
3. Button "Filters"
4. expand dropdown menu for Mode

**Expected Behavior**
I have QSOs in several different modes in my log, mostly digital modes, but also some phone contacts in SBB (USB)
But when I try to filter after used mode in order to find those USB QSOs the mode is not available for selection in the dropdown menu.
I've found the same issude in Cloudlog as well, so wavelog might have inherited it.

**Screenshots**
[image]

**Desktop Information:**
- OS:               Windows 11 Pro
- Browser:         Firefox
- Version:          132.0.2

Best 73 and keep up the good work!
Sebastian DL2KSE

AndreasK79, 2024-11-17:
The reason is to merge usb and lsb into ssb, but it has a side effect. I will make a fix and let you know when you can test!

AndreasK79, 2024-11-18:
@sebket if you run dev branch, you can pull the latest changes and test. If not, wait for the upcoming release.

sebket, 2024-11-18:
@AndreasK79 thanks for fixing this so rapidly. I'm running only one productive instance, so I'll wait for the upcoming release. Would it help if I'll give feedback here afterwards?

AndreasK79, 2024-11-19:
@sebket no problem. Any feedback is good, so let us know when you it's tested out.

sebket, 2024-12-05:
Just installed the new major release and checked right away wether searching for SSB (USB) QSOs is working now and yes, it works just fine! Thanks for the good work! 73 de Sebastian DL2KSE

PR #1216 (2024-11-18): [Advanced Logbook] Removed the if check to merge USB/LSB/SSB
To avoid USB/LSB/SSB not showing up in the filter dropdown, depending on how your mode/submode are logged.

This should fix #1213
closed 2024-12-05 by sebket