wavelog/wavelog #1233: QSO time wrong (UTC vs Local?)

pomtom44, 2024-11-19:
**Describe the bug**
Not sure if this should be a bug, or if its an issue with my setup, so happy to move to a discussion if required.

Brand new install, timezone is set properly.
However when going to log a QSO, it is showing what I believe is UTC time
It will show the correct time for a second, then change to UTC

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:              Ubuntu server 24.04
- PHP:            8.3.6
- MySQL/MariaDB:    5.7.33-0ubuntu0.16.04.1-log
- Wavelog:         1.8.8 

**Steps to Reproduce**
Checked server - reporting as local time
Checked settings - timezone set ok
Checked Config - Tried with both UTC and Local in the config file

**Screenshots**
[image]
[image]
[image]
[image]

HB9HIL, 2024-11-19:
Issue will be that Wavelog is uploading the QSOs to external services like QRZ.com, LoTW and so on directly from the database. One of the key features of Wavelog. But all data have to be in UTC. 

Currently there is not feature where Wavelog will store the QSO's in UTC but displays in Local Time. I'm sorry.

HB9HIL, 2024-11-19:
We could create a feature request from that but I can't tell you yet when and if this will be addressed

pomtom44, 2024-11-19:
That I understand, It would be handy to have a local time reference, so I don't have to do the conversions myself.
You already have the timezone stored for the user, so it might be a simple change of displaying local time rather than UTC time in the logbook.
Im not great with coding but I could throw together a proof of concept for review

HB9HIL, 2024-11-19:
> That I understand, It would be handy to have a local time reference, so I don't have to do the conversions myself. You already have the timezone stored for the user, so it might be a simple change [...]

I fear it won't be that easy. But sure, if you got a POC why not

phl0, 2024-11-20:
PR would be preferred :-)

pomtom44, 2024-11-22:
Ok so its the case of "The last 10% takes 90% of the work"
And of course the fun "Timezone conversions"

can I ask where the numbering and naming of the timezones comes from? as they don't seem to follow and standards that I can find 
I can't find the ID's listed anywhere public, so I assume it was just created in order when creating the database? but there are ones missing from the table so not sure where they go?

and the name doesn't follow any standards I can find
EG
"Pacific/Auckland" rather than "Auckland, Wellington"

pomtom44, 2024-11-22:
Also before I go messing with the timezone table in the database, is that data being used anywhere else?
I have had a quick look but nothing is jumping out at me obvious

pomtom44, 2024-11-24:
Sorry I must be doing something wrong.
When I try to make a PR is says I don't have access.
This is the first PR i am doing on github, so not sure what im doing wrong.

phl0, 2024-11-25:
> PR would be preferred :-)

I just had a look and a few more thoughts about the PR. In my opinion this should not be integrated. There are various reasons why hamradio logs are done in UTC one of them being the hassle to deal with different timezone offsets and daylight saving options. After viewing the PR I would refrain from merging this. Sorry for that.

closed 2024-11-25 by pomtom44