wavelog/wavelog #1238: Timeline Query times out

jwussler, 2024-11-21:
I am on the dev branch 1.8.8

When i run a query under analytics and select all and use the check mark for just lotw confrontations the page turns white and times out

[image]

I can run the query per band and it works fine. Just not when everything is selected to all

Joe
WAØO

int2001, 2024-11-21:
Hi @jwussler ,
can you provide some more information?

at first pse take a look at the JS-Console (F12) and network. Would be great if you can share the result of the call to `/timeline` when submit was pressed.

at 2nd: can you enable logging and show us whats in the application-log? (`/application/logs/`)

jwussler, 2024-11-21:
here the log file 

<?php defined('BASEPATH') OR exit('No direct script access allowed'); ?>

ERROR - 2024-11-21 07:27:19 --> Severity: error --> Exception: FUNCTION wavelog.0and does not exist /var/www/html/system/database/drivers/mysqli/mysqli_driver.php 301

[image]

int2001, 2024-11-21:
huh? That 0and thing is very strange.
can you provide an ADIF-Export? You can see my eMail-Adress at qrz.com

Idea: You imported an ADIF which was created with another tool. And there are Bands within that adif, which do not exist. but just an idea.

73 de DJ7NT

int2001, 2024-11-21:
tnx for the File.
Found the bug - it's not your ADIF which causes the problem.
i am going to fix it

int2001, 2024-11-21:
should be fixed now.

are u using docker (in that case i've to rebuild the dev-image - takes about 10minutes)?
otherwise simply pull latest dev

jwussler, 2024-11-21:
I am not on docket

PR #1240 (2024-11-21): Added whitespace
Fixes #1238 

a simple whitespace was forgotten. That leads to a 500 if someone did:
- Analytics
- Timeline
- All Bands / All Modes
- LoTW checked
closed 2024-11-21 by jwussler