wavelog/wavelog #1255: Can the Add Band/Mode to Favs function be designed to quickly reach the specified frequency?

XiangJiang-Wang, 2024-11-23:
Wavelog is not very compatible with the radio hardware interface of the radio station at present. I have been studying it for a long time but it has not been able to realize the linkage with the radio station. Every time I record a QSO, I need to manually fill in the frequency, which feels cumbersome. The collection function of the QSO page can only record the band and mode at present. It is recommended to add the frequency at the same time. I think most people will only communicate on a few commonly used frequencies, so I hope that the collection function can be designed to quickly reach the specified mode, band, and frequency.

HB9HIL, 2024-11-23:
Hello @XiangJiang-Wang 

The favourite feature in the QSO logging also stores the frequency. But due it's design it only can store one frequency per band/mode combination. 

What exactly is your issue / What exactly do you want to do? Store multiple Frequencys on the same Band/Mode?

XiangJiang-Wang, 2024-11-23:
I would like to be able to store multiple frequencies in the same band/mode

int2001, 2024-11-25:
Reason for your request is that you don't have CAT?
If using CAT (e.g. via WavelogGate or other 3rd party-tools), mode/band/qrg is filled automatically.

The FAV-Feature was built initially for SAT-QSOs. And the 100% correct QRG plays no role when obtaining Awards, doing LoTW, eQSL and so on.

i'd like to understand your use-case a bit more.,

Even if its possible to store multiple-frequencies within that Fav-Feature: What are you going to do? Create a fav for every QRG? perhaps i am missing something here.

XiangJiang-Wang, 2024-11-25:
I continue to be able to prepare to enter frequencies while recording qso, so I hope that this collection function can quickly reach the specified mode/band/frequency.

int2001, 2024-11-27:
to be honest: i still don't get it.

Examples:
a)You're working with CAT - so no favourite needed (i think you can confirm that - because QRG/Mode will be instantly overwritten by CAT)

b) You're "in the field" with no CAT. You turn over the Band and answer a QSO. In that case you don't know the QRG/Mode before, because it is always different. Simply enter the QRG/Mode and thats it.

c) You're "in the field" with no CAT. You're doing "running" (means: You call CQ on a fixed QRG). You enter the QRG/Mode ONCE and log all your contacts.

d) You're doing SAT-OPs. You have to enter correct prop-mode (SAT), correct SAT_MODE (e.g. V/U), correct QRGs for up/down and at least correct Mode. All this has to be done ONCE. But it's a lot. This is where favs came into the game.

The favs are curently saving the Band/Mode, QRG and SAT settings.

i still don't know what you're trying to do.

i try to construct an example:
You want Favs for every possible QRG on the 40m Band? in that case you'd have a huge Dropdown and it's much easier to enter just the right QRG. You can even enter the QRG with the Unit (e.g. 14.190 MHz instead of 14190000)

Please help us to understand.

closed 2025-01-28 by github-actions[bot]