wavelog/wavelog #1262: Mode Changes Unexpectedly When QSYing to Call Sign

spREGR, 2024-11-25:
Describe the bug
When I click on a call sign in the Bandlist section, the radio QSYs (changes frequency) to the desired call sign, but the mode changes from CW-U (Continuous Wave Upper) to USB (Upper Sideband) unexpectedly.

Version Information

    OS: [Debian 12]
    PHP: [8.3.12]
    MySQL/MariaDB: [11.3.2-MariaDB-1:11.3.2+maria~ubu2204]
    Wavelog: [1.8.8]

Steps to Reproduce

    Go to Bandlist in Wavelog
    Click on a desired call sign
    The radio QSYs to the selected call sign
    Observe that the mode switches from CW-U to USB

Expected Behavior
I expect the radio to QSY to the selected call sign, but the mode should remain as CW-U instead of changing to USB.

Radio: FT991A
Interface between wavelog and radio : Wavelog-gate version v.1.0.16
Interface between Wavelog-gate and radio: FLrig 2.0.04
[image]

73
VU2TUM

int2001, 2024-11-25:
pse update to latest WavelogGate - there's a new checkbox/option in it, which prevents changing of the mode:

[image]

spREGR, 2024-11-26:
Hi, I updated the wavelog-gate to latest version and the Set MODE via FLrig is checked. Still, the mode changes from CW to LSB or USB

int2001, 2024-11-26:
na, uncheck it and save it.
Needs to be unchecked!

spREGR, 2024-11-27:
Thanks, Works now.

closed 2024-11-27 by spREGR