wavelog/wavelog #1269: UI Bug regarding QSL status

HB9HIL, 2024-11-27:
### Discussed in https://github.com/wavelog/wavelog/discussions/1267

<div type='discussions-op-text'>

<sup>Originally posted by **gen-intel** November 27, 2024</sup>
Switching between QSL info when you are in Live QSO does not show the individual QSL status. It seems that QSL Card "sticks" to   the selection. i.e when you select LoTW you still see the Card and then on the bottom the LoTW info. This behavior is different in Live QSO compared to looking at a QSO's QSL's when in the "Logbook"

Is this by design?

It's a bit confusing.
 
[image]
[image]
[image]
[image]
[image]
[image]
</div>

gen-intel, 2024-11-28:
So far so good.
Thank you.
N5HC

HB9HIL, 2024-11-28:
Bug fixed. Thx for the report @gen-intel

closed 2024-11-28 by HB9HIL