wavelog/wavelog #1324: User distance setting set to miles but overview distance column shows kilometers

krisp, 2024-12-05:
I am using the latest version of the container under docker compose, which reports version as 1.8.8 in the debug menu. 

**Steps to Reproduce**
Account -> Wavelog preferences
  Select Distance as a logbook column. 
  Select Miles as measurement distance.
  Scroll to bottom and click save.
Return to Logbook->Overview, observe distance column in km.

**Expected Behavior**
I expect this column to be displayed in miles.

**Additional Context**
I also noticed the account date format does not carry over to the Past QSO entry page which is where it is most needed but I found a separate bug for this and it looks like a more complicated issue.

HB9HIL, 2024-12-15:
Hello @krisp 

this one was fixed last week and will be in the next release.

WrathPak, 2024-12-20:
Hi there, Sorry I made a new issue. I figured I would come here and let you know I can confirm I am on 1.9 and this issue still persists.

PR #1331 (2024-12-07): [Distance unit] Dashboard and logbook overview shows distance in set unit
Fixed #1324
closed 2024-12-29 by github-actions[bot]