wavelog/wavelog #1341: No paper QSL if I have another QSO with paper QSL already sent

DK9JC, 2024-12-10:
I like to send out paper QSL cards but I want to avoid sending multiple QSLs for the same band or satellite to people I have in my log. Especially on LEO satellites and QO-100 I have some people in my log 30 times or more.

So it would be good if I could set a checkbox somewhere “Avoid dupe QSL” or something like that, so that only those who have not yet confirmed another QSO on the band or on the satellite (SAT name) can be seen in the Advanced log under QSL filters "QSL sent" requested or no.

The benefit is not only there for every other user, it also helps us to save resources and paper.

HB9HIL, 2024-12-15:
Hello @DK9JC 
there is a draft in #1353 but there are some open questions. Tbh I don't see a "Avoid dupe QSL" feature as there are so many cases and combinations where you may want to send a QSL card, even a qsl card was already sent to that call/band/mode. 

Therefore I added a column in the QSL queue to indicate if a qsl card was already sent to this call/band/mode combination. But I will need to dig into the code again to see if it's not too restricting and take  a look at database performance.

int2001, 2024-12-15:
Hi Johnny :)
Additional question:
If you're working mobile/roving, i assume that you want to send a card (because it's a new grid). Right?
So imho the station_location should be as well part of the dupe-check.

i don't know if this is an edge-case and is also relevant for the HF-People (i think yes - IOTA/POTA/SOTA/etc.)

So the main-question is:
What exactly is a dupe? 
- Already confirmed band/mode?
- Already confirmed band/mode/location?
- Already confirmed band/mode/location and REF of the QSO-Partner?
- Another combination?

We don't want to complicate it, but i fear there are a lot of different views on "what is a QSL-Dupe".

DK9JC, 2024-12-15:
> Hello @DK9JC there is a draft in #1353 but there are some open questions. 

Hello, Fabian. Thank you.

> Tbh I don't see a "Avoid dupe QSL" feature as there are so many cases and combinations where you may want to send a QSL card, even a qsl card was already sent to that call/band/mode.

The optimal feature should be really a “**hard condition**” and **not** every possible case should be considered. Of course, it could be that someone requests a QSL again, but then I can still do it manually.

A “hard feature" would help me to cope with the masses of QSOs. If one does have many QSOs in a log, he can of course do it all manually.

> Therefore I added a column in the QSL queue to indicate if a qsl card was already sent to this call/band/mode combination. 

The display is good, but if I then have to click a thousand individual checkboxes, it doesn't help much in the end and I would rather either not send any more cards from me or continue to send them to everyone, including the dupes.

> Hi Johnny :) 

Hello Jörg.

> Additional question: If you're working mobile/roving, i assume that you want to send a card (because it's a new grid). Right?

No, not for roving "only Grids". It is not manageable. Of course if I get a request I send out a card, but then I can manually check the QSO. When I am in another DXCC (e.g. when I was in Iceland) I send out a QSO to everyone via DARC ADIF print.

> So imho the station_location should be as well part of the dupe-check.

Yes, but for me I would be good to check DXCC-wise and not for Grid locations (I have 80 station locations already...)

> i don't know if this is an edge-case and is also relevant for the HF-People (i think yes - IOTA/POTA/SOTA/etc.)

I dont send out a paper card for every SOTA / POTA / Roved SAT Grid as it is not manageable with many thousand QSO. Therefore they can be treated as "dupe", if SAT Name or Band is the Same. 

> We don't want to complicate it, but i fear there are a lot of different views on "what is a QSL-Dupe".

I can totally understand it and also to keep it simple and understandable.

A dupe is a dupe if within my own same DXCC a contact on the same band or same SAT has already received a paper QSL.

The same SAT is still to be discussed. I would even say: A QSL for any SAT is good enough. Anyone who would like to have a QSL for another SAT can send me a card and will then receive a card back for the QSO with the respective (new, other) SAT name. Then it is uniform and the same conditions apply for SAT as for direct connections.

Thank you both for reading and investing your time to make wavelog a better logger. :-)

int2001, 2024-12-16:
i managed to optimize the query for the last-qsl-count. Still not very happy with it, but it is in progress and doesn't slow down the whole log any longer.

pse check the PR against your base-PR, @HB9HIL - https://github.com/HB9HIL/wavelog/pull/122

closed 2025-07-17 by github-actions[bot]