wavelog/wavelog #1347: Suggestion: Show Download Progress for eQSL Cards

xyz667, 2024-12-13:
**What Feature is Missing?**
I suggest some sort of visual indicator that downloads are in progress when downloading eQSL cards (Third-Party Services - eQSL Import/Export - Download eQSL cards).
Maybe sth like in the "update from Callback" function when updating multiple QSOs? There, the remaining updates are counted down, which could be adapted to a countdown of eQSL cards remaining to be downloaded.

**Why is This Feature Important?**
Downloading eQSL cards (Third-Party Services - eQSL Import/Export - Download eQSL cards) can take quite some time, currently without any indication to the user that something is happening. There is no progress bar or any other visual cue.

**Is This Feature Personal or Beneficial to Others as Well?**
I think this feature would benefit any eQSL user.

int2001, 2024-12-16:
We're going to disable the complete Massdownloader (one can re-enable it with a config.php - switch).
Reasons for that:
- You can download eQSLs with nearly instant Feedback via the "arrows" at Logbookadvanced/Overview. A once viewed image is still saved on your server. So 2nd time viewing the image will be much faster.
- The whole massdownloader-Feature isn't threadsafe. means: If u click a few times on "Download all" it is able to crash your server.
- The function is redundant to the already mentioned viewer at the arrow.

Sorry. But the effort to design it safe is to big, and there are usable alternatives for it.

xyz667, 2024-12-17:
@int2001 would it be much effort to create a loop that (if I may put it that way in non-coder-speak) clicks all the „view/download“ buttons on that page one by one? (=perfoms the eQSL sync just like a press on that button, and then loop though the whole table of un-synced eQSLs, and abort if any error occurs)
IMO this is exactly the suggestion, and should be thead-safe if a single download action is thread-safe?

xyz667, 2024-12-17:
Addition: should there be no function to download all un-synced eQSL cards anymore we (still) would need a way to check which cards are not downloaded, right? Otherwise missing cards would go unnoticed.

int2001, 2024-12-17:
You can view the QSL-Cards by clicking on the green arrow.
There's even a carousel-function for clicking them one by one.

I don't get the point where massdownloading helps you? There's no function within WL to download all cards at once.

what am i missing?

xyz667, 2024-12-17:
@int2001 well currently there still is such a function (Third-Party Services - eQSL Import/Export - Download eQSL cards).
The „arrow“ in the logbook you mention does not indicate if the card has already been downloaded or if download is outstanding. And I would need to hover over *all* the green arrows in my whole(!) logbook to check, even if that indication would be added to the tooltip. This is unworkable, especially as there already is functionality to show the not yet downloaded eQSL cards, in Third-Party Services - eQSL Import/Export - Download eQSL cards.

I get that the *current* *download* function on that page seems to be problematic and needs to be reworked/removed.

I don‘t get why it would be desirable to scap the whole functionality of showing unsynced cards (if this is what you are suggesting above, not sure if I got the PR code right?).

And maybe there even is a low-effort fix to the current mass-downloader. We already have the table of un-synced cards in cards Third-Party Services - eQSL Import/Export - Download eQSL cards. As I said before, couln‘t there be a smarter way to download the missing cards on the basis of that table content - one by one (but automatically)?

int2001, 2024-12-17:
Sorry, Question was not answered: why do you need Wavelog to download all cards at once?

Showed you a way to view them. Very convenient. You don't need to "hover" over each arrow. Simply click the 1st arrow and browse through it with the carousel-function.

Tbh: the whole image-function is based on an undocumented API / a sort of Webscraper for eQSL. Not reliable. 

Of course you're free to create a PR for the requested feature. It's open source...

int2001, 2024-12-17:
i think we're tlaking about different things here.
let me explain:

The standard eQSL-Sync marks ALL QSOs with eQSL-Cards with a green down arrow, but it doesn't download the card.
Thats important for understanding.

if you want to view a card, you can do this even for old QSOs.
You can see the "fresh" eQSL-confirms at the dashboard. if you receive a eQSL for a very old eQSL, it'll be shown there.
Furthermore you're able to use the Logbook-Advanced to view them.

Again the question, because you mentioned again the issue "...make sure all cards are downloaded...".
Why do you want Wavelog to Download all the cards at once. This question is still open.
They're downloaded on demand (when viewing them) and cached. No reason to DL them all.

What do you do with a collection, where you're only to access them one by one?? Wavelog is a online-Tool. Without the 3rd party-service you won't even get the information about a received QSL.

to be clear: we don't remove the feature. we just disable it. a experienced user (like you are) may re-enable it by a config-option. No need to put more effort into this feature, because:
a) undocumented API
b) needless (see argumentation over and over above) Feature. every card can be viewed and will be cached within WL
c) Not threadsafe
d) Securityhole for DoS-Attacks

xyz667, 2024-12-17:
@int2001 just to explain why I‘m being such a pain:  ;-)

I download and store all eQSL cards, and would love to have an easy way of doing so.

Why would I want to download every single eQSL card?
Because I don‘t want to be reliant on 3rd parties. eQSL might shut down their service. They might get ransomwared like the ARRL, but never recover. They might have technical problems just as I need the card. They might close their API. My Internet connection fails for prolongued time. Whatever the reason.
With all cards downloaded and stored locally forever I can‘t be bothered.

Your alternative download ways do not seem like viable solutions to the use case described above.
- On the dashboard I can only see today‘s new eQSL confirmations. Would I have to login at 23:59 UTC every day to download all cards?
- Via the logbook and carousel it‘s a pain to make sure *everything* is downloaded.
- Generally downloading (and storage/caching) is no explicit function but just a necessity in the *viewing* process - except in the now disabled part.

Yes, the functionality to see all un-synced cards is still there (when enabled in config).
I just worry that on the day after tomorrow or so it will get removed „as nobody uses this anyway“.

Trust me, I‘d love to contribute. But unfortunately have no clue about php.

And I still can‘t believe I‘m the only one with the „get all and store locally“ mentality. :-)

int2001, 2024-12-17:
ok. sorry - i don't think we'll find a common view on that here.

Yes, the cards are stored within Wavelog. But there's no function available to download them to your local computer. So the only way to access the cards is viewing them step by step - i described that way several times.

There's also **absolutely no way** to make the current function for mass-downloading safe/secure or adding a progressbar.

I suggest you re-enable the existing feature in the next version to use the "Mass-Downloader" on _your own risk_.

You pointed us to a security-hole/design-failure. The only responsible thing we can do is to disable that "feature".

Wait - one more thing: since eQSL doesn't provide an API for mass-downloading the images in a safe way, it **may** be a chance to request such an API there or address the request there.

to be honest: The whole eQSL-Thing is fallen out of time. It doesn't count for any Award (except their own ones).

i hope you understand our move. Security/Safety is important for wavelog/us.

Best 73, DJ7NT

xyz667, 2024-12-17:
@int2001  just a correction: of course I don’t want downloads to my local computer, but to my Wavelog instance. Just to be independent from eQSL. 

I get that the mass-downloader is problematic. 

But why kill the viewing of undownloaded eQSL cards when trying to solve that problem?

int2001, 2024-12-17:
Nobody talked about disabling the whole page where you can see the "undownloaded images".
It's just about the button at the end of that page. The mass-dl-feature.

xyz667, 2024-12-17:
@int2001  then I’m somewhat relieved 
worst case all the buttons on that page had to be pressed. (if the config option isn’t set)