wavelog/wavelog #1348: Contest Logging Live: Different Exch / Serial needed

DK9JC, 2024-12-13:
Tomorrow is ARRL 10m Contest and was thinking to use Wavelog for logging.

According to the Rules, USA stations (also XE/VE) sending their state as exhange. E.g. 59 CT. Stations not in USA/XE/VE sending a serial number. So would be nice, if it can be set in wavelog accordingly.

Now as is I can chose "Exhange" **for both** or "Serial" **for both** but not Exchange for them and counting n+1 serial for me.

From the ARRL rules:

"Exchange: All stations send a signal report. W/VE stations, including Alaska and Hawaii) send their state
or province. (District of Columbia stations send “DC”.) Mexican stations send their state or province. DX
stations send a serial number. Maritime mobile stations send ITU Region (1-3)"

[image]

[image]

HB9HIL, 2024-12-13:
Simply set "Serial + Exchange" and leave fields empty you don't need.

HB9HIL, 2024-12-13:
PS: In "More Settings" you can adjust the sequence of the fields if needed

DK9JC, 2024-12-13:
> Simply set "Serial + Exchange" and leave fields empty you don't need.

Thanks Fabian, I will try. 

But it would be better if it would change accordingly. Maybe even with some dependency. If my call does not come from the USA/VE/XE, it could be set automatically maybe that I only give out serial and only receive State/Provice from the other.

(Yes I know Wavelog is not a full contest-logger...)

DK9JC, 2024-12-15:
Can be closed @HB9HIL 

> Simply set "Serial + Exchange" and leave fields empty you don't need.

Worked.

Thank you.

closed 2024-12-15 by DK9JC