wavelog/wavelog #1354: Problem with Contest Logging Live (portable out in the field)

DK9JC, 2024-12-15:
Yesterday I was out in the field to take part in a contest. At some point the QSO on the contest page (lower part of the contest screen) disappeared. 

The serial numbers kept counting up (+1), but there was no more dupe warning, because the dupes are based on the QSOs displayed on the contest page. there was no way to get these QSOs “back” into the live contest. I then searched in a 2nd tab in the normal log each time to see if the QSO was already in the log.

Control + F5 did not solve the problem, another contest restart was needed, so I entered the last serial number as the start. The Cabrillo export on the other hand exported my 3 started sessions correctly in one.

Just a note for the backlog if you should tackle the issue again 

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Apache
- PHP:             8.2.26-nmm1 
- MySQL/MariaDB:   10.11.8-MariaDB
- Wavelog:         1.9

**Desktop Information:**
- OS:               Windows 10
- Browser:          Firefox latest build

DK9BT, 2024-12-30:
i had a similiar issue.....Contest Log was gone...

You can reproduce it by logging in wavelog demo instance...start a live contest log and then click 2x the page refresh button

HB9HIL, 2024-12-30:
> i had a similiar issue.....Contest Log was gone...
> 
> You can reproduce it by logging in wavelog demo instance...start a live contest log and then click 2x the page refresh button

I will check that ASAP.

risototh, 2025-01-19:
Same issue here... First time using the wavelog, directly in contest... First few QSO also skipped some serial numbers, so the log will be a bit messy... :D So for now I'm back on paper :D

jax-mckenzie, 2025-01-19:
Can confirm odd behavior from the contest logger. Worked for a while and showed contacts but now it doesn't even have my contest selected anymore. Not sure if it auto-resets after a day or something but it was the january VHF contest which lasts longer than 24 hours. There's no serial number just grid exchange so not really a huge deal, i'll just export an adif of all the contacts during that period, but clearly not working correctly.

In addition, contest logger suffers from issue #1339 which I was going to post a new issue for.

PA2EJD, 2025-02-09:
Same issue here, lost contact with wifi which seemed to empty the list of stations already worked in the contest. Unable to recover list so no dupe checking. Had to use a second tab in browser to manually check for dupes. Log was complete no data lost just annoying during the contest.

AndreasK79, 2025-02-18:
I did push a small fix regarding this the other day, so it might have solved the issue with the session being gone. The code is in the dev branch.

DK9JC, 2025-02-22:
> I did push a small fix regarding this the other day, so it might have solved the issue with the session being gone. The code is in the dev branch.

Thank you! 

Sadly as a /p station I am not a big contester, so can not check in "real" life soon, but I will give it a try at some point. Maybe someone else can test it out and report back.

closed 2025-09-09 by github-actions[bot]