wavelog/wavelog #1359: Error message clarification request: You could automate this process by making it a cronjob.

kmanwar89, 2024-12-18:
Hi,

Just started using Wavelog and I love it. I'm trying to figure out how to automatically backup my QSO's as an ADI file.

When I go to the "backup" menu, below is the message that is displayed.

[image]

However, it doesn't give any guidance on *how* a user might create said cronjob. I'm familiar with modifying the crontab and administering Linux machines, so that isn't the knowledge gap here - I'm mostly asking how I would do the custom cronjob *specifically as it relates to Wavelog*.

Is it at possible to add in the functionality to create the cronjobs in the UI, or even just provide the user guidance on which file/where to edit to add in? I personally use a tool like [crontab guru](https://crontab.guru/) to create cronjobs for testing before deployment.

Thanks in advance for any guidance you can provide, and 73!

HB9HIL, 2024-12-19:
Our Wiki has a search box where you can search for "Backup"

https://github.com/wavelog/wavelog/wiki/Backup

kmanwar89, 2024-12-20:
> Our Wiki has a search box where you can search for "Backup" 😃
> 
> https://github.com/wavelog/wavelog/wiki/Backup

That's what I get for not searching :sweat_smile:  thank you for pointing me in the right direction!

HB9HIL, 2024-12-22:
You reopened. Something open?

kmanwar89, 2024-12-22:
Apologies - I had started typing an issue out but realized it was just me skipping steps :) Disregard!

closed 2024-12-22 by kmanwar89