wavelog/wavelog #1365: Logbook front page of wavelog shows distances in km when settings set to miles

WrathPak, 2024-12-20:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
I have set up my preferences to show distanes in miles. 

[image]

I see distances in kilometers from the front page of wavelog on the logbook below the map. 

[image]

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- Wavelog:          1.8.8

**Steps to Reproduce**
1. Set distance preferences in wavelog to miles, and have distances set as a column in your logbook
2. go to wavelog front page where map and recent qsos are
3. see that the distances are in kilometers on the logbook

**Expected Behavior**
Describe what you expected to happen.

phl0, 2024-12-20:
You should update your instance. This has been implemented in version 1.9:
https://github.com/wavelog/wavelog/pull/1331

WrathPak, 2024-12-20:
@phl0 I did just update. I am on 1.9, issue persists.

WrathPak, 2024-12-20:
I tried switching distance preferences, then switching back. did not help. still in km. Also tried turning off the distance column on the logbook, then turning it back on. no luck. still shows km. I am definitely on 1.9

AndreasK79, 2024-12-20:
This was fix right after 1.9 release: https://github.com/wavelog/wavelog/pull/1331
Either switch to dev branch, or wait for the new release.

closed 2024-12-20 by AndreasK79