wavelog/wavelog #1383: Allow API Endpoints to take input as url parameters instead of raw data.

ethancedwards8, 2024-12-26:
**What Feature is Missing?**
To use the API endpoints, data must currently be passed through mechanisms similar to `curl -d {...}`. Instead, we should be able to use `log.com/api/get_contacts_adif?key=XXXX&station_id=X&fetchfromid=XXX`. (or whatever endpoint). 

**Why is This Feature Important?**
This would make development easier and would allow more flexibility in the APIs usage. 

**Is This Feature Personal or Beneficial to Others as Well?**
I think it would help developers make software that interacts with wavelog. I am interested in building a mobile app that would allow you to sync QSOs between wavelog and the app, and having this interface to the API would help.

**Implementation**
This file seems to be where the information is stored: https://github.com/wavelog/wavelog/blob/c0fd01e0010fd63fde9efd4efd341df34e11fba1/application/controllers/Api.php

This tutorial does a good job at explaining how to parse arguments from the URL: https://www.geeksforgeeks.org/how-to-get-parameters-from-a-url-string-in-php/ .

I will attempt to work on this in the coming weeks. I will implement it for all endpoints.

filipmelik, 2024-12-30:
Hi, just my thought, so please take it with a grain of salt, as I am not the maintainer 😊 

What kind of problem are you trying to solve here? You are describing that the reason is developing the new mobile app and to make things easier - I honestly see the point here. Sending POST request is not much of a difference than sending a GET request. Of course, constructing the URL string might be easier a little bit, but I do not personally think it is a justification for creating duplicates of all endpoints to be accessible as GET. What I see is potential of a lot of added code in the codebase, that needs to be maintained over time and that is my concern here. 

I do not know all the API endpoints wavelog currently have, but also please note, that not all data is suitable for sending via the GET method (array of structured objects for example) and also query string in GET requests are limited to certain length that vary depending on the web server serving the wavelog website. 

Maybe I would reconsider the ` I will implement it for all endpoints` part - If we are talking RESTful API here, the GET endpoints should be getting data, POST endpoints should be updating data etc. So there might be usecase for making some POSTs also as GETs, however, I am not sure the other way around for the reasons outlined above 😊 

Again, this is just my personal opinion here, the core maintainers have the final word here of course 👍

int2001, 2024-12-30:
tbh: We live in the year 2024.
All GET-Data isn't secure (we know, we're using it as well, and many other Ham-Tools also).

What exact problem are you facing with POST?
Perhaps we can help/assist you, or give you a hint how to implement POST into your Application?!

ethancedwards8, 2024-12-30:
@int2001 and @filipmelik 

I'm not sure how y'all got the impression that I wanted to move everything to GET. That is not what I intended. The existing endpoints will stay and I will not change them.

All this issue mentions is simply adding the ability to use URL parameters instead of data to send the requests (whether that be GET, POST, or otherwise). Of course we wouldn't move everything to GET, it wouldn't make sense. Again, I'm not sure how I gave off that impression, but that is not at all what I was going for.

HB9HIL, 2024-12-30:
> @int2001 and @filipmelik
>
> I'm not sure how y'all got the impression that I wanted to move everything to GET. That is not what I intended. The existing endpoints will stay and I will not change [...]

What exactly do you mean with "URL Parameters"? Can you make an example?

HB9HIL, 2024-12-30:
Because you posted this link in your initial request: https://www.geeksforgeeks.org/how-to-get-parameters-from-a-url-string-in-php/

This describes indeed parsing "URL parameters" in PHP. BUUUTTTT:

1.  This is a GET Request. In GET requests the transmitted data is written in the URL. Like here `name=Tonny`
```
https://www.geeksforgeeks.org?name=Tonny
``` 

2. The link describes in fact how to parse "URL Paramaters" (aka. GET Parameters) in plain PHP. We are using the Codeigniter Framework. While you could parse GET data aswell in this format. It's better to use the Codeigniter syntax here which would be: 

```
$name = $this->input->get('name');  // parses  the 'name' as GET parameter in the request data
```

Currently we are parsing the required data as POST data which is definitly the better way here:

```
$name = $this->input->post('name');  // parses  the 'name' as POST parameter in the request data
```

---

Fact is: What you describe here as "URL Parameters" is a GET request `log.com/api/get_contacts_adif?key=XXXX&station_id=X&fetchfromid=XXX`

int2001, 2024-12-31:
so, we can close this one, @ethancedwards8 after @HB9HIL clarified that you meant a GET-Request?
if you've problems to implement a proper POST to your 3rd-party let us know. perhaps we're able to help.

ethancedwards8, 2024-12-31:
Yes. Thanks for your help.

closed 2024-12-31 by ethancedwards8

HB9HIL, 2024-12-31:
> Yes. Thanks for your help.

You're welcome :)