wavelog/wavelog #1394: POTA References Aren't Imported when Importing from ADI files

ethancedwards8, 2024-12-26:
**Describe the bug**
When attempting to upload a POTA activation the QSOs do not retain POTA reference information for either operators (activator (me) or hunter for p2p contacts)

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Fedora Linux + Docker
- PHP:              8.3.15
- MySQL/MariaDB:    11.3.2-MariaDB-1:11.3.2+maria~ubu2204
- Wavelog:          1.9 (dev branch)

**Steps to Reproduce**
1. Perform POTA activation in Hamrs
2. export ADI file
3. Import file to wavelog
4. check QSO log. POTA references are not there.

**Expected Behavior**
POTA references should be imported form ADI files. Specifically my_sig_info (activator) and sig_info (hunter) fields.

**Screenshots**
Include screenshots if applicable to illustrate the issue.

**Desktop Information:**
- OS:               MacOS
- Browser:        Brave

phl0, 2024-12-26:
This is a known issue. Wavelog uses the POTA specific ADIF fields for that information. But up to now even POTA itself does not make use of those and stil relies on SIG amd SIG_INFO. From my point of view POTA and corresponding software should make use of the specifically foe them introduced fields here.

ethancedwards8, 2024-12-26:
What do you suggest users do in the meantime?

phl0, 2024-12-27:
> What do you suggest users do in the meantime?

I will look into it if we implement a (temporary) workaround.

phl0, 2025-01-02:
I had a play with hamrs. To be honest the export from the app is a bit confusing:

```
<ADIF_VER:5>3.1.4
<CREATED_TIMESTAMP:15>20250102 214915
<PROGRAMID:5>HAMRS
<PROGRAMVERSION:6>2.25.6
<EOH>

<RST_RCVD:2>59
<RST_SENT:2>59
<CALL:6>DN1YWF
<OPERATOR:5>DF2ET
<FREQ:2>14
<BAND:3>20m
<MODE:3>SSB
<TX_PWR:2>10
<POTA_REF:7>DE-0795
<COMMENT:4>test
<TIME_ON:6>214857
<QSO_DATE:8>20250102
<NAME:7>Florian
<QTH:11>Tippelsberg
<STATE:2>NW
<GRIDSQUARE:6>JO31om
<SIG_INFO:7>DE-0795
<MY_SIG_INFO:0>
<EOR>
```

The POTA ref of the contacted station is included as POTA_REF. This is correctly parsed by Wavelog. The POTA ref is also listed in SIG_INFO but without SIG being set. The data is redundant here and leading to double entry of the POTA reference unter POTA_REF and SIG_INFO:

[image]

So I do not see a thing that we could change here. Feel free to give a more detailed description of what fails in your case.

The MY_ fields are not parsed on ADIF upload by the way. Instead those are set per station location for each location/reference.

ethancedwards8, 2025-01-02:
Interesting. Mine doesn't export POTA_REF. Were you using HAMRS or HAMRS Pro? I use regular HAMRS (the old version, both are free). 

If you had success using HAMRS pro, I will switch to it.

phl0, 2025-01-02:
Downloaded and tested with vv2.25.6 from https://hamrs.app/...

phl0, 2025-01-02:
It is not the pro version if I see it correctly.

ethancedwards8, 2025-01-02:
All v2 versions are Hamrs Pro. Plus, I tested it with Hamrs Pro and it exported the POTA_REF field. So, that appears to be the difference.

No problem, I can use Pro from now on. Thanks for looking into this.

closed 2025-01-02 by ethancedwards8