wavelog/wavelog #1395: SOTA Operator References Aren't Imported From ADI files

ethancedwards8, 2024-12-27:
**Describe the bug**
When importing SOTA logs, the activator SOTA reference is not imported. However, the hunter references are. 

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Fedora Linux + Docker
- PHP:              8.3.15
- MySQL/MariaDB:    11.3.2-MariaDB-1:11.3.2+maria~ubu2204
- Wavelog:          1.9 (dev branch)

**Steps to Reproduce**
1. Perform SOTA activation in Hamrs
2. export ADI file
3. Import file to wavelog
4. check QSO log. Activator SOTA references are not there.

**Expected Behavior**
SOTA references should be imported form ADI files. Specifically my_sota_ref (activator) field.

**Desktop Information:**
- OS:               MacOS
- Browser:          Brave

**Additional Context**
Unlike POTA, SOTA does use its own ADI fields like sota_ref and my_sota_ref.

HB9HIL, 2024-12-27:
MY_SOTA_REF does get imported. But We override it with the SOTA Reference in the Station Location. When importing a new SOTA activity, you have to create a new station location for that with the referring details like SOTA, POTA and so on.

ethancedwards8, 2024-12-27:
If that is the case, shouldn't the third party services in Station Location (the QRZ, Clublog, HRDLog integrations) be moved to account settings instead?

HB9HIL, 2024-12-27:
@ethancedwards8 

No because the most services are based on the used Callsign. For example  the QRZ.com Integration requires the QRZ logbook api key which is bind to the callsign of that logbook in QRZ.com (You can't use the QRZ API Key for HB9HIL if the Station Callsign is HB9HIL/P)

Therefore the most integrations are based on the Station Callsign -> on the Station Location. There are exceptions for this like LoTW and eQSL, which requires the Login Details in the User Settings. eQSL needs only the QTH Nickname in the Location and LoTW does the callsign declaration based on the LoTW Certificates. 

You see there is not "that one way". Different services require different credentials on different station/callsign details... Thats the nature of hamradio 😆

ethancedwards8, 2024-12-27:
I see. Thank you for explaining this to me. I will close this issue.

closed 2024-12-27 by ethancedwards8

HB9HIL, 2024-12-27:
> I see. Thank you for explaining this to me. I will close this issue.

You're very welcome. Don't hestitate to ask in the [Discussions](https://github.com/wavelog/wavelog/discussions) if something is not clear :)