wavelog/wavelog #1396: eQSL Card Search is case sensitive and doesn't recognize 0 in place of Ø

ethancedwards8, 2024-12-27:
**Describe the bug**
eQSL Card Search doesn't not recognize 0 in place of Ø and is case sensitive.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Fedora Linux + Docker
- PHP:              8.3.15
- MySQL/MariaDB:    11.3.2-MariaDB-1:11.3.2+maria~ubu2204
- Wavelog:          1.9 (dev branch)

**Steps to Reproduce**
1. Go to Logbook, eQSL cards.
2. Search callsign without uppercasing

**Expected Behavior**
calls to return regardless of 0 or upper/lower case

**Additional Context**
Maybe to fix this everything typed can automatically be converted to uppercase and 0 replaced like the general log screen does

int2001, 2024-12-31:
Fixed exactly that way. tnx for headsup, @ethancedwards8

ethancedwards8, 2025-01-01:
My only concern with how this was implemented is that you can see the letters being retyped/the 0 being replaced whereas on the header when you type the correct case/0 is automatically inputted.

ethancedwards8, 2025-01-01:
Here is a demonstration

https://github.com/user-attachments/assets/57949fec-7ead-4fbd-a522-bf9b2775a378

ethancedwards8, 2025-01-07:
@int2001

closed 2025-01-16 by github-actions[bot]