wavelog/wavelog #1403: Saved Preferences on Advanced Logbook Page

ethancedwards8, 2024-12-27:
**What Feature is Missing?**
Advanced logbook preferences (like station location, # of results, etc) should be saved. Each time I go off the page these preferences reset. 

**Why is This Feature Important?**
It is annoying and error prone to have to set all the preferences each time.

**Is This Feature Personal or Beneficial to Others as Well?**
This feature will be beneficial to all concerned.

ethancedwards8, 2024-12-28:
I'd be happy to do the legwork on this PR, I'm just not sure where to start.

HB9HIL, 2024-12-28:
I justed discussed this with @AndreasK79. One way would be to store the current values in `localStorage`. Would be reset on page clearing Browser Cache or hit "Reset". 

Tbh I don't see this stored in database. Only in browser. 

Questionable is: Store everything? Like
also filters or only certain stuff. 

Implementation won't be that hard but we should think that through.

ethancedwards8, 2024-12-29:
Good question. I'm not sure. Definitely the number of QSOs and the station location at a minimum. Maybe not filters. 

Hopefully others will provide some feedback.

AndreasK79, 2024-12-29:
Never thought about adding preferences to the advanced logbook, but I don't see why not.
For now, I guess #results and locations are enough. I really don't see the point in storing filters, as that usually changes.

If you would like to try to add this, go ahead. As @HB9HIL mentioned, use localStorage to save this.

ethancedwards8, 2025-01-30:
Solved by: https://github.com/wavelog/wavelog/pull/1576

closed 2025-02-05 by ethancedwards8