wavelog/wavelog #1406: The latest streak is not shown under "Days with QSOs" - "Streaks"

sebket, 2024-12-29:
This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area

Describe the bug
The latest streak is not shown under "Days with QSOs" - "Streaks"

Version Information
We need some information about your versions. You can get all Information in the Admin > Debug Menu

OS: Ubuntu Server 24.04.1 LTS
PHP: 8.3.6
MariaDB: 10.11.8
Wavelog: 1.9

Steps to Reproduce

Do a streak longer than any previous one and then check whether it shows up.

Expected Behavior
I just had QSOs 5 days in a row. The QSOs are in the log and during the streak there was the message under "Days with QSOs" - "Streaks" that informed me about the current streak.
Now the streak is over (ended 2 days ago) but it doesn't show up in the list.
It should show up because it is my longest streak yet.

Desktop Information:

OS: Windows 11 Pro
Browser: Firefox
Version: 132.0.2

Best 73,
Sebastian DL2KSE

AndreasK79, 2024-12-29:
Seems I may have been able to reproduce the bug here. I could get it to happen if no QSOs are logged after the streak has ended. Is this the case with your log?

sebket, 2024-12-29:
Yes, that is the case.
Streak started December 23rd and ended December 27th, so two days ago. No more QSOs since.

sebket, 2024-12-29:
Just created a dummy QSO with today's date and yes, this triggers the streak to be shown as expected.
As soon as I delete that dummy QSO again the streak also disappears.

AndreasK79, 2024-12-29:
Yeah, that was the same I found. I could not reproduce, until I removed the latest QSO. I'll take a look and see if I can figure out what is going on. I haven't touched the streak code in years :)

Thanks for reporting!

sebket, 2024-12-29:
Well, I take it as a hint that it's time for some real QSOs to enter the log today!
Jokes aside, thanks for having a look at this.

AndreasK79, 2024-12-30:
@sebket fix is in latest dev.

sebket, 2024-12-30:
Okay, thanks!
I'll wait for the next release to try it out if that's all right.

sebket, 2025-01-02:
Seems to work as expected now, thank you!

PR #1409 (2024-12-30): Streak fix
Fixes #1406.
closed 2025-01-02 by sebket