wavelog/wavelog #1436: After updating to 1.9.1 I can't open any QSO to view its details

sebket, 2025-01-02:
Describe the bug
I just updated my installation from 1.9 to 1.9.1 via the update function under "Debug Information".
The update went smoothly and everything seems to work okay, but I can't open any QSO to view.
When I click on the callsign simply nothing at all happens.
I can open the QSO for editing, though.

Version Information
We need some information about your versions. You can get all Information in the Admin > Debug Menu

OS: Ubuntu Server 24.04.1 LTS
PHP: 8.3.6
MariaDB: 10.11.8
Wavelog: 1.9.1

Steps to Reproduce
I don't know... be me, update from 1.9 to 1.9.1 and try to open any QSO for viewing after that.

Expected Behavior
QSOs should open to be able to view them just like they did before the update.

Desktop Information:

OS: Windows 11 Pro
Browser: Firefox
Version: 132.0.2

Best 73,
Sebastian DL2KSE

phl0, 2025-01-02:
Did you try a reload without caching? E.g. CTRL + F5? What happens if you click to open a QSO? Does the browser console show any errors?

sebket, 2025-01-02:
Reload without caching doesn't change anything.
I've also tried it in Private Mode or with another browser with which I've never accessed wavelog before, all with the same result.

When I click to open a QSO nothing at all happens. The browser console in multiprocess mode is showing a handful of errors about cookies (language, ci_wavelog) that do not habe a proper "SameSite" attribute.
Those errors appear in the moment I click on a QSO, but also periodically without clicking on anything.

phl0, 2025-01-02:
Which page are you calling the QSO details from?

sebket, 2025-01-02:
I've tried them all: start page, logbook overview and logbook advanced, none works.
What's supposed to be opened seems to be:
javascript:displayQso(713)

phl0, 2025-01-02:
That looks correct. Please have a look at the network tab of browser console to see if there is a HTTP request triggered.

sebket, 2025-01-02:
I think I was able to fix it.
I download the new release as ZIP, deleted the \application\views folder from the server and copied it back from the extracted ZIP.
That solved it, it's working now. Perhaps the update routine got confused as I had some additional .bak files in some folders from trying out some changes in the code of some PHP files.
So, mea culpa! Won't put any additional files in there anymore.
Thank you very much for your help anyways!

sebket, 2025-01-02:
There might have been a PHP file in there somewhere with a line or two added or removed, yes.
But location and/or filenames were not altered.

phl0, 2025-01-02:
Closing here then since we cannot debug the root cause for the issue :)

closed 2025-01-02 by phl0