wavelog/wavelog #1439: Callbook should be a User setting

KidVizious, 2025-01-02:
I am running a Wavelog server and wanted to provide access to some of my friends. Unfortunately, since the callbook credentials are global, all of their lookups will also be using my Callbook credentials.

Callbook login creds should be moved into user settings so that each user can use their own accounts for callbook lookups.

HB9HIL, 2025-01-02:
Since this is only for the Callbook Lookup and has no affect to your user account,
why does it matter?....

KidVizious, 2025-01-02:
It does have an impact on my user account.  Assume I'm not a paid QRZ subscriber, then I'm limited in the number of lookups per day. If multiple people are using my login for lookup, that limit will be reached quite quickly. 

And even if I am a paid subscriber, it just seems to me that in general requiring a user to use personal credentials as a global server setting isn't best practice.

HB9HIL, 2025-01-03:
> It does have an impact on my user account. Assume I'm not a paid QRZ subscriber, then I'm limited in the number of lookups per day. If multiple people are using my login for lookup, that limit will be reached quite quickly.

If you don't have a paid subscription at QRZ.com this will also reduce the amount of data you get back from QRZ.com. For example you won't get gridsquares.  

> And even if I am a paid subscriber, it just seems to me that in general requiring a user to use personal credentials as a global server setting isn't best practice.

Well it's up to you to share your Wavelog Instance with other users. You also could set up hamqth or qrzcq as a callbook source. 

In my eyes it's much much easier and better practice to purchase the subscription and share the costs among the users (if the money is an issue) than to force every user to use a paid subscription individually if they want to full joy of qrz.com data...

int2001, 2025-01-03:
Full ACK to  @HB9HIL's argumentation.

We have discussed this feature request in our team and have come to the conclusion that it will **not be implemented** in wavelog.

Lets summarize it - You've a couple of options:
1.) Purchase a subscription.
2.) Switch over to another callbook-provider like hamqth.
3.) Stop sharing your own subscription and leave it blank. - nobody forces you to share your instance.

However. Many thanks for the idea. We're happy about every feedback. But not every wish can be fullfilled.

If you're extremly uncomfortable with this decision, you can still add the feature on your own to your instance. That's one of the benefits a FOSS-Software allows.

closed 2025-01-09 by int2001