wavelog/wavelog #1446: Missing Clublog Upload/Download Page in Third Party Services Drop Down

ethancedwards8, 2025-01-03:
**What Feature is Missing?**
I noticed that the Clublog Upload/Download Page was missing from the Third Party Services column.

**Why is This Feature Important?**
Having it would allow the manual triggering of uploading/downloading clublog confirmations.

**Is This Feature Personal or Beneficial to Others as Well?**
It would be beneficial to all concerned.

int2001, 2025-01-03:
Yep. this is indeed missing.
Would be happy if u can start the implementation for it.

AndreasK79, 2025-01-08:
- [x] Massedit Clublog status at LBA with #1476

AndreasK79, 2025-01-10:
- [x] Edit Clubslog status at Edit-QSO with #1489

ethancedwards8, 2025-01-10:
Also, clublog needs to be added to the QSO search page.

[image]

int2001, 2025-01-10:
is already implemented:
enable Clublog at Settings, and it'll show up there.

[image]

ethancedwards8, 2025-01-10:
Ahhhh. Thanks.

AndreasK79, 2025-01-20:
I am working on a GUI page for Clublog upload/download. It will come when it's done.

AndreasK79, 2025-01-31:
GUI is merged into dev.