wavelog/wavelog #1449: Frames arround the xOTA & Co. Fields

dg2ron, 2025-01-03:
I am a big fan of xOTA & Co. collections. 
Now when I hear an activation, I first have to check whether I have already worked on the reference. This is very time-consuming with many references.
If I don't do that, I don't want to wait in the pileup for 5-10 minutes only to find out that I already have everything in the logbook.
It would actually be nice if the frame came up when I had entered all the numbers. 
Maybe RED = ATNO and GREEN = confirmed or worked or "in the Log"

dg2ron, 2025-01-03:
this is what a mean

[image]

int2001, 2025-03-27:
i think we can close this now, with the enhancement introduced in #1790

closed 2025-03-27 by dg2ron