wavelog/wavelog #1456: Could not migrate database. Ajax crashed. Status: 0 Status Text: error

M0LTE, 2025-01-04:
**Describe the bug**

If you change the URL during installation from an IP to the system's hostname, the database migrations fail to run with `Could not migrate database. Ajax crashed. Status: 0 Status Text: error`

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Debian 12 / 	Apache/2.4.62
- PHP:             8.2.26
- MySQL/MariaDB:   10.11.6-MariaDB-0+deb12u1
- Wavelog:       1.9.1

**Steps to Reproduce**
1. Install new instance (I used https://community-scripts.github.io/ProxmoxVE/scripts?id=wavelog)
2. Change URL during setup from http://10.1.1.1 (for example) to http://wavelog (for example) - where "wavelog" *does* resolve to 10.1.1.1
3. Continue with installer
4. Observe failure

```
 Wavelog Installer Debug Log
-------
Log Location: ../application/logs/installer_debug.log

INFO - 2025-01-04 12:10:16 --> Called Installer index.php
INFO - 2025-01-04 12:10:16 --> With URL: http://10.45.0.206/install//
INFO - 2025-01-04 12:10:16 --> From IP: 10.45.0.171
INFO - 2025-01-04 12:10:16 --> Set language cookie to "English"
DEBUG - 2025-01-04 12:13:29 --> Start writing config.php
INFO - 2025-01-04 12:13:30 --> Data validation passed.
INFO - 2025-01-04 12:13:30 --> CI_ENV is not set. Using default config.php config path.
INFO - 2025-01-04 12:13:30 --> config.php template file read successfully.
INFO - 2025-01-04 12:13:30 --> Config.php file prepared successfully. Writing to file...
INFO - 2025-01-04 12:13:30 --> config.php file written successfully.
DEBUG - 2025-01-04 12:13:30 --> File: config.php successfully written
DEBUG - 2025-01-04 12:13:30 --> Start writing database.php
INFO - 2025-01-04 12:13:31 --> Data validation passed.
INFO - 2025-01-04 12:13:31 --> CI_ENV is not set. Using default database.php config path.
INFO - 2025-01-04 12:13:31 --> database.php template file read successfully.
INFO - 2025-01-04 12:13:31 --> Database config file prepared successfully. Writing to file...
INFO - 2025-01-04 12:13:31 --> database.php file written successfully.
DEBUG - 2025-01-04 12:13:31 --> File: database.php successfully written
DEBUG - 2025-01-04 12:13:31 --> Start creating database structure with assets/install.sql
INFO - 2025-01-04 12:13:31 --> Data validation passed.
INFO - 2025-01-04 12:13:31 --> SQL queries prepared successfully. Writing to database...
INFO - 2025-01-04 12:13:32 --> Database tables created successfully.
DEBUG - 2025-01-04 12:13:32 --> Tables successfully created
DEBUG - 2025-01-04 12:13:32 --> Start migrating database to the newest version.
ERROR - 2025-01-04 12:14:00 --> Could not migrate database. Ajax crashed. Status: 0 Status Text: error
```

HB9HIL, 2025-01-04:
Can you explain "Change URL during setup"

HB9HIL, 2025-01-04:
In your install debug log I can see that your wavelog instance is at `http://10.45.0.206/install/` 

So http://wavelog has to resolve to 10.45.0.206

M0LTE, 2025-01-04:
> Can you explain "Change URL during setup"

During setup there is a page which allows you to set the URL Wavelog will use. It defaults to http://ipaddress.

> So http://wavelog/ has to resolve to 10.45.0.206

It does, hence the bug report.

HB9HIL, 2025-01-04:
okay. We fixed an issue here in `dev` branch. If you want to run the development branch it should work, otherwise you may wait for the next release. But this issue should be already fixed

M0LTE, 2025-01-04:
Thanks. I just went with IP address and I will reconfigure manually myself. But future me will be eternally grateful.

HB9HIL, 2025-01-04:
To adjust the address you can simply change "base_url" in your config.php. But this behaviour is already fixed in dev why I mark it as ready 2 close

welcome 2 wavelog :)

HB9HIL, 2025-01-04:
Should be fixed by https://github.com/wavelog/wavelog/pull/1444

M0LTE, 2025-01-04:
Yep, thanks for that, and thanks for the fix.

closed 2025-01-13 by github-actions[bot]