wavelog/wavelog #1463: Add bearing/beam heading to list of colum options for overview page and include in QSO popup when clicking map

krisp, 2025-01-05:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
I'm not sure if I missed an option for this or if something else is required to record/retrieve heading information other than four/six digit grid square. I notice the antenna analytics page always shows a round circle as if it isn't getting headings for any qsos. My request is the ability to choose a bearing/heading column for the overview page and also be able to see the heading to a station when clicked on the map. The popup currently includes date, band and mode. Perhaps an option to also display distance and heading in the popup? 

**Why is This Feature Important?**
This feature would help with VHF/UHF weak signal where beam width is narrow and not every station's location is committed to memory. I tend to work the same two dozen or so stations within a five or six grid radius of my QTH and it would be great to see headings at a glance in the overview window, or in the advanced search where the column also appears to be unavailable. The only place I can currently find heading is in the Add to Log window after filling in grid (as expected) and on the grid square map when clicking. I often end up on QRZ to confirm headings and it would be nice to use the logbook instead, especially if I already have their six digit grid logged.

**Is This Feature Personal or Beneficial to Others as Well?**
I would think this feature would be useful to others interested in confirming or looking up headings for remote stations and would speed the lookup of beam headings, and the heading could potentially used by a rotator add-on down the road.

This is not something that is high priority or need but would certainly be a nice-to-have.

int2001, 2025-01-09:
You mean sth. like this at Overview and Dashboard?
[image]

BTW: Bearing and distance is shown in the Map at Logbook-Advanced

int2001, 2025-01-09:
implemented now (in dev):
- Map-information (see screenshot)
- configurable bearing-col for Dashboard and Overview.

krisp, 2025-01-09:
Yes that is exactly what I mean. Looks good! thank you.

closed 2025-01-18 by github-actions[bot]