wavelog/wavelog #1464: Club station page missing

hellofinch, 2025-01-06:
Hello,
 I translate on the Weblate. I see club station including. How can I visit these pages? I need some context to translation. I try to review the code. But failed to match on the Wavelog pages.
 Is it correct?

73
: )

int2001, 2025-01-06:
Good morning,
if you want to test Clubstations you could(!) install latest dev-Version.
It is currently in development and not yet released.

pse be aware of the fact that dev may contain bugs (dev is for development and testing)

closed 2025-01-06 by int2001

hellofinch, 2025-01-06:
Thanks for your reply. I installed the dev version. Can you tell me how to find the club station pages?
[image]
: )

int2001, 2025-01-06:
you have to enable it at config.php - see https://github.com/wavelog/wavelog/wiki/Clubstations#setting-up-a-clubstation

int2001, 2025-01-06:
you're welcome.
Would be great if you can test everything and give a feedback (not about possible missing features, e.g. Contest is disabled for club, but for bugs)

hellofinch, 2025-01-06:
OK！
73
: )