wavelog/wavelog #1471: Error while loading station setup page.

N3evin, 2025-01-07:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
After latest update 1.9.1, everytime i enter station setup, it just pop out saying `An error ocurred while making the request`

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Ubuntu 22.04.4 LTS
- PHP:              8.1.2-1ubuntu2.19
- MySQL/MariaDB:    5.5.5-10.6.18-MariaDB-0ubuntu0.22.04.1
- Wavelog:          1.9.1

**Steps to Reproduce**
1. Go to 'station setup'
2. Popout saying `An error ocurred while making the request`

N3evin, 2025-01-07:
```
<?php defined('BASEPATH') OR exit('No direct script access allowed'); ?>

INFO - 2025-01-07 01:48:11 --> Config Class Initialized
INFO - 2025-01-07 01:48:11 --> Hooks Class Initialized
INFO - 2025-01-07 01:48:11 --> UTF-8 Support Enabled
INFO - 2025-01-07 01:48:11 --> Utf8 Class Initialized
INFO - 2025-01-07 01:48:11 --> URI Class Initialized
INFO - 2025-01-07 01:48:11 --> Router Class Initialized
INFO - 2025-01-07 01:48:11 --> Output Class Initialized
INFO - 2025-01-07 01:48:11 --> Security Class Initialized
INFO - 2025-01-07 01:48:11 --> Input Class Initialized
INFO - 2025-01-07 01:48:11 --> Language Class Initialized
INFO - 2025-01-07 01:48:11 --> Config file loaded: /var/www/html/application/config/gettext.php
INFO - 2025-01-07 01:48:11 --> Loader Class Initialized
INFO - 2025-01-07 01:48:11 --> Config file loaded: /var/www/html/application/config/wavelog.php
INFO - 2025-01-07 01:48:11 --> Config file loaded: /var/www/html/application/config/bands.php
INFO - 2025-01-07 01:48:11 --> Config file loaded: /var/www/html/application/config/lotw.php
INFO - 2025-01-07 01:48:11 --> Helper loaded: url_helper
INFO - 2025-01-07 01:48:11 --> Helper loaded: security_helper
INFO - 2025-01-07 01:48:11 --> Helper loaded: language_helper
INFO - 2025-01-07 01:48:11 --> Database Driver Class Initialized
INFO - 2025-01-07 01:48:11 --> Session: Class initialized using 'files' driver.
INFO - 2025-01-07 01:48:11 --> Migrations Class Initialized
INFO - 2025-01-07 01:48:11 --> Language file loaded: language/english/migration_lang.php
INFO - 2025-01-07 01:48:11 --> Database Forge Class Initialized
INFO - 2025-01-07 01:48:11 --> Model "Options_model" initialized
INFO - 2025-01-07 01:48:11 --> Model "User_options_model" initialized
INFO - 2025-01-07 01:48:11 --> Model "Logbooks_model" initialized
INFO - 2025-01-07 01:48:11 --> Model "Stations" initialized
INFO - 2025-01-07 01:48:11 --> Model "Oqrs_model" initialized
INFO - 2025-01-07 01:48:11 --> Controller Class Initialized
INFO - 2025-01-07 01:48:11 --> Model "Cat" initialized
INFO - 2025-01-07 01:48:11 --> File loaded: /var/www/html/application/views/components/radio_display_table.php
INFO - 2025-01-07 01:48:11 --> Final output sent to browser
INFO - 2025-01-07 01:48:11 --> Total execution time: 0.0045
INFO - 2025-01-07 01:48:16 --> Config Class Initialized
INFO - 2025-01-07 01:48:16 --> Hooks Class Initialized
INFO - 2025-01-07 01:48:16 --> UTF-8 Support Enabled
INFO - 2025-01-07 01:48:16 --> Utf8 Class Initialized
INFO - 2025-01-07 01:48:16 --> URI Class Initialized
INFO - 2025-01-07 01:48:16 --> Router Class Initialized
INFO - 2025-01-07 01:48:16 --> Output Class Initialized
INFO - 2025-01-07 01:48:16 --> Security Class Initialized
INFO - 2025-01-07 01:48:16 --> Input Class Initialized
INFO - 2025-01-07 01:48:16 --> Language Class Initialized
INFO - 2025-01-07 01:48:16 --> Config file loaded: /var/www/html/application/config/gettext.php
INFO - 2025-01-07 01:48:16 --> Loader Class Initialized
INFO - 2025-01-07 01:48:16 --> Config file loaded: /var/www/html/application/config/wavelog.php
INFO - 2025-01-07 01:48:16 --> Config file loaded: /var/www/html/application/config/bands.php
INFO - 2025-01-07 01:48:16 --> Config file loaded: /var/www/html/application/config/lotw.php
INFO - 2025-01-07 01:48:16 --> Helper loaded: url_helper
INFO - 2025-01-07 01:48:16 --> Helper loaded: security_helper
INFO - 2025-01-07 01:48:16 --> Helper loaded: language_helper
INFO - 2025-01-07 01:48:16 --> Database Driver Class Initialized
INFO - 2025-01-07 01:48:16 --> Session: Class initialized using 'files' driver.
INFO - 2025-01-07 01:48:16 --> Migrations Class Initialized
INFO - 2025-01-07 01:48:16 --> Language file loaded: language/english/migration_lang.php
INFO - 2025-01-07 01:48:16 --> Database Forge Class Initialized
INFO - 2025-01-07 01:48:16 --> Model "Options_model" initialized
INFO - 2025-01-07 01:48:16 --> Model "User_options_model" initialized
INFO - 2025-01-07 01:48:16 --> Model "Logbooks_model" initialized
INFO - 2025-01-07 01:48:16 --> Model "Stations" initialized
INFO - 2025-01-07 01:48:16 --> Model "Oqrs_model" initialized
INFO - 2025-01-07 01:48:16 --> Controller Class Initialized
INFO - 2025-01-07 01:48:16 --> Model "Cat" initialized
INFO - 2025-01-07 01:48:16 --> File loaded: /var/www/html/application/views/components/radio_display_table.php
INFO - 2025-01-07 01:48:16 --> Final output sent to browser
INFO - 2025-01-07 01:48:16 --> Total execution time: 0.0063
INFO - 2025-01-07 01:48:21 --> Config Class Initialized
[... 84 lines cut ...]
INFO - 2025-01-07 01:48:22 --> Config file loaded: /var/www/html/application/config/wavelog.php
INFO - 2025-01-07 01:48:22 --> Config file loaded: /var/www/html/application/config/bands.php
INFO - 2025-01-07 01:48:22 --> Config file loaded: /var/www/html/application/config/lotw.php
INFO - 2025-01-07 01:48:22 --> Helper loaded: url_helper
INFO - 2025-01-07 01:48:22 --> Helper loaded: security_helper
INFO - 2025-01-07 01:48:22 --> Helper loaded: language_helper
INFO - 2025-01-07 01:48:22 --> Database Driver Class Initialized
INFO - 2025-01-07 01:48:22 --> Session: Class initialized using 'files' driver.
INFO - 2025-01-07 01:48:22 --> Migrations Class Initialized
INFO - 2025-01-07 01:48:22 --> Language file loaded: language/english/migration_lang.php
INFO - 2025-01-07 01:48:22 --> Database Forge Class Initialized
INFO - 2025-01-07 01:48:22 --> Model "Options_model" initialized
[... cut ...]
INFO - 2025-01-07 01:48:22 --> Model "Stations" initialized
INFO - 2025-01-07 01:48:22 --> Model "Oqrs_model" initialized
INFO - 2025-01-07 01:48:22 --> Controller Class Initialized
INFO - 2025-01-07 01:48:22 --> Helper loaded: form_helper
INFO - 2025-01-07 01:48:22 --> Model "User_Model" initialized
INFO - 2025-01-07 01:48:22 --> Model "Logbook_model" initialized
INFO - 2025-01-07 01:48:22 --> Final output sent to browser
INFO - 2025-01-07 01:48:22 --> Total execution time: 0.1678
```

With full logs.

HB9HIL, 2025-01-07:
Hello @N3evin 

In the log is nothing which could lead to this error. The `An error ocurred while making the request` is made by an ajax request from your browser to your server. Can you please check out your browsers developer console and see if any errors there.

N3evin, 2025-01-07:
[image]

Just did, don't see anything in console log.

N3evin, 2025-01-07:
Found the issue, does not seems to be 1.9.1 issue. But I had enable `compress_output`, disable it sees to solve the issue

closed 2025-01-07 by N3evin