wavelog/wavelog #1491: BandMap DataTables warning

gen-intel, 2025-01-12:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
BandMap: DataTables warning: table id=DataTables_Table_0 - Requested unknown parameter '6' for row 0, column 6. For more information about this error, please see https://datatables.net/tn/4.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu

Server Software | nginx/1.26.2
PHP Version | 8.3.6 OK
MySQL Version | 10.11.8-MariaDB-0ubuntu0.24.04.1
Codeigniter Version | 3.2.0-dev

**Steps to Reproduce**
1. Go to Tools -> Bandmap
2. See error
3. Click OK until all entries are "OK"
4. BandMap shows DX Cluster entries
5. Next update start showing the same Warning message.

**Expected Behavior**
Describe what you expected to happen.

**Screenshots**
Include screenshots if applicable to illustrate the issue.

**Desktop Information:**
- OS:               macOS 15.2 (24C101)
- Browser:         Safari
- Version:          18.2 (20620.1.16.11.8)

**Smartphone Information:**
- Device:           [e.g., iPhone6]
- OS:               [e.g., iOS8.1]
- Browser:          [e.g., stock browser, safari]
- Version:          [e.g., 22]

**Additional Context**
Provide any other relevant context about the problem.

int2001, 2025-01-13:
Did you press CTRL+F5 (Hard reload) for the page?

gen-intel, 2025-01-13:
No.

But I just cleared the browser history and that cleared the issue.

Maybe CTRL+F5 do the same thing.

All good now.

73 de N5HC

/Conny

On Jan 13, 2025, at 12:55 AM, Joerg (DJ7NT) ***@***.***> wrote:

Did you press CTRL+F5 (Hard reload) for the page?

closed 2025-01-21 by github-actions[bot]