wavelog/wavelog #1505: Callbook search not working, says Username / password required

WEKarnesky, 2025-01-15:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
Everytime I look up a callsign, the callbooks section just reports "Username / password required"

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Apache/2.4.62 (Debian) (Docker container on Synology)
- PHP:              8.3.15
- MySQL/MariaDB:    11.3.2-MariaDB-1:11.3.2+maria~ubu2204
- Wavelog:          1.9.1

**Expected Behavior**
I would expect information to appear from QRZ.

**Screenshots**
Example lookup:
[image]
showing that I am requesting pictures from callbook:
[image]
Showing QRZ is selected
[image]
Showing QRZ username and password is set
[image]

**Desktop Information:**
- Browser:         Chrome
- Version:          131.0.6778.205

**Additional Context**
During set up I configured Callbook download to use QRZ and entered username/password.  Thinking it was an issue of using 2fa, I disabled 2fa.  Still didn't work.  Then thinking it was a bad password I went in to the settings file (located at `/docker/wavelog/config/config.php`) and C&Ped the username and password into a browser to confirm that it works.  I then tried buying a qrz subscription (which I was interested in anyway) to try there, and it still isn't working.

Docker Compose File:
```
---
services:
  wavelog-db:
    image: mariadb:11.3
    container_name: wavelog-db
    environment:
      MARIADB_RANDOM_ROOT_PASSWORD: yes
      MARIADB_DATABASE: REDACTED
      MARIADB_USER: REDACTED
      MARIADB_PASSWORD: REDACTED# <- Insert a strong password here
    volumes:
      - /volume1/docker/wavelog/db:/var/lib/mysql:rw
    restart: unless-stopped

  wavelog-main:
    container_name: wavelog-main
    image: ghcr.io/wavelog/wavelog:latest
    depends_on:
      - wavelog-db
    environment:
      CI_ENV: docker
    volumes:
      - /volume1/docker/wavelog/config:/var/www/html/application/config/docker:rw
      - /volume1/docker/wavelog/uploads:/var/www/html/uploads:rw
      - /volume1/docker/wavelog/data:/var/www/html/userdata:rw
      - /volume1/docker/wavelog/logs:/var/www/html/application/logs
    ports:
      - "8086:80"
    restart: unless-stopped

```

HB9HIL, 2025-01-15:
Quick question to proof my suspicion:

Did you edit the config.php directly in the docker container? To be precise: Did you edit the file `/var/www/html/application/config/config.php`?

WEKarnesky, 2025-01-15:
> Quick question to proof my suspicion:
> 
> Did you edit the config.php directly in the docker container? To be precise: Did you edit the file /var/www/html/application/config/config.php?

The only time I edited the file was by going to the folder outside of docker.  I then restarted the container after editing.  The username and password and QRZ setting were already there when I opening it.  The only change I made was fullname = true.

HB9HIL, 2025-01-15:
In a docker setup the file `/var/www/html/application/config/config.php` is fully ignored. 

Please check the config file in your docker volume 
in your case: `/volume1/docker/wavelog/config/config.php`

WEKarnesky, 2025-01-15:
That is the file I am editing.  The screenshots are from `/volume1/docker/wavelog/config/config.php`

HB9HIL, 2025-01-15:
are there more then 2 files in the folder `/volume1/docker/wavelog/config`?

I expect there to be only

config.php
database.php

WEKarnesky, 2025-01-15:
Correct.
[image]

HB9HIL, 2025-01-15:
[image]
I just tested our official `docker-compose.yml`-Template and it works as expected with my QRZ credentials

Do you have any special characters in your password? Looking for single `'` or double quotes `"`

WEKarnesky, 2025-01-15:
> Do you have any special characters in your qrz password?

There is a dollar sign `$` as part of the password

WEKarnesky, 2025-01-15:
> Could you perform a Qrz XML check here?
> 
> https://xml.qrz.com/xcheck

It failed, but I'm not sure why.

Here are the steps I took:

1. I copied the User name from the config file and pasted it in
2. I copied the password from the config file and pasted it in
3. Clicked Login -- **Resulted in Password Incorrect**.
4. I went to _QRZ.com_ in a private browsing tab
5. Manually entered the same username
6. Copied the password into the field
7. Clicked Login -- **Results in Successful Login**

> Sorry for asking so much questions. But I'm likely sure that the issue is something in your configuration. But as we always try to make things intuitive this should be more easy to solve. Trying to find the issue now

I appreciate the detailed debugging, no need to apologize.

WEKarnesky, 2025-01-15:
I'm changing my password to remove any any all special characters and see if that resolves the issue.

HB9HIL, 2025-01-15:
> It failed,

Voilà. So it's likely an issue with your QRZ.com account. The message "Username / password required" is as far as I know a response from QRZ which is only returned in the UI. 

Are you sure you entered the exact username? Did you double check that? Since you wrote in 5 that you manually entered the username but in 1 you copied the username from the config.php

WEKarnesky, 2025-01-15:
Success!

Resolution notes:

First, the Dollar sign **May or May Not** have been a problem.  It seem the XML portal sends the password as part of the url, and there are a number of other characters that cause issues with parsing there.

Second: While QRZ.com allows EITHER the Email or Callsign, The XML portal ONLY allows the callsign, I had been using Email.

Thanks for you assistance!

[image]

closed 2025-01-15 by WEKarnesky

HB9HIL, 2025-01-15:
Great! Thanks for your patience and cooperation :)

Welcome to Wavelog :)