wavelog/wavelog #1508: DXCC country update failure.

K9bdh, 2025-01-15:
Morning,  Got everything running but updating the DXCC country list always fails.  It just spins and spins and then finally crashes.   everything on the debug screen is green  Is there something I could be missing?

k9bdh

AndreasK79, 2025-01-17:
Hi, can't really say why have that trouble, but you can try the dev branch. I tweaked the DXCC update so it should use less resources and time. Perhaps that will help you.

K9bdh, 2025-01-17:
Thanks

I took this approach  which is not a good one but I figured I fat fingered something so I re installed the cloud server and did it step by step again and cranked the memory to 512 meg and all that jazz and appears to all be working now.   Next to get my Mac sir to pad to wave log. :). I think I am going to like tis software.  Thanks a bunch

closed 2025-01-28 by github-actions[bot]