wavelog/wavelog #1523: Merge QSO fields for duplicate QSOs

ethancedwards8, 2025-01-19:
**What Feature is Missing?**
The ability to merge fields from duplicate QSOS:

Example:
You have two logs with duplicate QSOs they share all the fields exactly except one field
Log 1 has:
sig_info:us-3378
Log 2 has:
pota_ref:us-3378

Log 1 was already imported several weeks ago. The user would like to upload log 2 and have the qso in wavelog reflect the new field from log 2. So, when log 2 is uploaded, the pota_ref field should be added to the qso.

**Why is This Feature Important?**
It would better flesh out the system and allow users to update their logs.

**Is This Feature Personal or Beneficial to Others as Well?**
This would help everyone moving from Hamrs to Hamrs Pro or some other log where they recently added support for pota_ref and my_pota_ref fields.

phl0, 2025-01-20:
Same opinion as @int2001. The effort of coding would be huge and I don't see too many people needing this. Agree that it is a pity with POTA and the various ADIF fields. But I am afraid we cannot help here. Sri

ethancedwards8, 2025-01-20:
No problem. I understand.

Would there be any possibility of a feature that would allow you to simply copy one field to another? For example, copy the sig_info field to pota_ref? That way you can do it all in lba + I could potentially see this feature being easier to implement/more useful for others.

closed 2025-01-26 by ethancedwards8