wavelog/wavelog #1524: Add a button to move from callsign search page to Live QSO, copying relevant data

TomJohnson0358, 2025-01-20:
When listening for stations, I prefer to use Wavelog to do the initial callsign search.  When I make the contact, I would like to be able to click a button, which would take me to the "Live QSO" page, with the callbook detailed information displayed along with the callsign.

This would save time by not having to change the page and retype in the callsign. It would also avoid having to use QRZ for search in another window, and also avoid copy/paste of the callsign.

int2001, 2025-01-20:
A question of Worflow.
my worflow:

i always have "Live QSO" open.  when listening, i type in the call and see who it is. if(!) i work the station, i press save. if i don't work it: Reset

how are you doing it? Search is for search, not for preparing the QSO.
QSO can be either prepared in the way is described or by clicking on the Call while in Cluster (Partner is automatically transferred to Live-QSO-Window)

TomJohnson0358, 2025-01-20:
OK, I guess the problem is that I have done the search in the live qso panel, then hit "return" before filling anything in the other fields.  That action commits the QSO with incomplete or incorrect information.

So, I stay away from "Live QSO" until I have actually made contact.

So I suppose I need to remember to use Tab.

Sorry.

closed 2025-01-20 by TomJohnson0358

int2001, 2025-01-20:
Pro-Tip: Space works also