wavelog/wavelog #1534: Add state/province information to Live QSO and Post QSO pages?

TomJohnson0358, 2025-01-21:
**What Feature is Missing?**
Please add the state or province data from the callbook data populated into Live QSO and Post QSO pages.

**Why is This Feature Important?**
This would avoid having to visit QRZ or other callbooks to see the state/province.  Thus I would know to initiate a call for a new state or province

HB9HIL, 2025-01-21:
Hello @TomJohnson0358 

afaik this is already working. But most of the people do not set state/province in the callbook... 

You can verify with my call HB9HIL. It shouöd change the state (in this case it's called canton) to "Appenzell Innerrhoden (AI)"

TomJohnson0358, 2025-01-21:
Thanks for the quick reply! 

I see no state/canton here.  Screenshot attached, as well as screenshot from debug page.

[image]
[image]

TomJohnson0358, 2025-01-21:
For what it is worth, my own state does not appear, but my city and county do appear.  Screenshot attached.

[image]
[image]

QRZ data attached as well.

HB9HIL, 2025-01-22:
switch to tab "General"

TomJohnson0358, 2025-01-22:
I am sorry. Closing

closed 2025-01-22 by TomJohnson0358

HB9HIL, 2025-01-22:
Glad you found it :)