wavelog/wavelog #1552: Updating QSL information in one click/one go (QRZ XML subscriber)

PE1PQX, 2025-01-25:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
I find that updating QSL information in a QSO-log very usefull. I would like to see a possidility top update all QSO's QSL information in one go ("Update QSL information of all QSO's"). As a QRZ XML subscriber I find it usefull

**Why is This Feature Important?**
I don't think it is important, only very useful after setting up Wavelog and/or uploading a big load of QSO's.

**Is This Feature Personal or Beneficial to Others as Well?**
I think it is beneficial for other users also, updating QSL info can help others to administrate their QSL distribution. 
Also if QSL info state 'no paper QSL', it could help identify these QSO's for not sending a paper QSL.

AndreasK79, 2025-01-25:
You mean like in the Advanced Logbook? Mark QSOs -> Actions -> Update from Callbook?

PE1PQX, 2025-01-25:
> You mean like in the Advanced Logbook? Mark QSOs -> Actions -> Update from Callbook?

I think I need to learn to look further... Again thank you Andreas!

AndreasK79, 2025-01-25:
No worries. Impossible to discover or know everything, and the documentation could always be better 😊

PE1PQX, 2025-01-25:
> No worries. Impossible to discover or know everything, and the documentation could always be better 😊

No arguments there... but looking around also can help to 'learn' WaveLog. (or whatever package you like to use)

closed 2025-01-26 by PE1PQX

NoobKeksTV, 2026-02-20:
> You mean like in the Advanced Logbook? Mark QSOs -> Actions -> Update from Callbook?

Sorry to open this again.
I know about this Feature but where would it pull the data from? Only from my alredy-made QSO or from some live data such as the QRZ XML API ? And where would i enter my details for it, i havent found a setting for that on the Page Settings or Logbook/Station Settings.