wavelog/wavelog #1556: Unable to complete install - "Could not migrate database. Ajax crashed. Status: 0 Status Text: error

TEBandCo, 2025-01-26:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
During first time install (http://fqdn/install or localIP/install), the installer fails at stage Running database migrations.
Error log "Could not migrate database. Ajax crashed. Status: 0 Status Text: error

[image]

**Version Information**

- OS:               Ubuntu 24.04.1 LTS
- PHP:              8.2
- MySQL/MariaDB:    mysql  Ver 8.0.40-0ubuntu0.24.04.1 for Linux on x86_64 ((Ubuntu))
- Wavelog:          1.9.1

**Steps to Reproduce**
1. New install
2. complete the /install/ wizard
3. get to installation, and wait for it to complete

**Expected Behaviour**
Complete the install 

**Screenshots**

[image]

**Additional Context**
I have attempted to complete the install with the FQDN and local IP.
I have used the FQDN and local IP in the `Website URL` section. 
Each time the database is dropped: 
[image]

This has been followed, however, the change reverts on each install run:

[image]

HB9HIL, 2025-01-26:
this is a known issue and should be already fixed in the `dev` branch. 

Hint about the config.php: You only can edit the file `application/config/config.php` AFTER the installation was successfull. Do not edit any other file

HB9HIL, 2025-01-26:
if possible please try again with the `dev` branch and do not edit any of the config.php before the installation was successful.

closed 2025-02-04 by github-actions[bot]

zhangyulu8023, 2025-04-17:
I met the same report,but i dont know how can i kill it