wavelog/wavelog #1585: Exception: Object of class WAE could not be converted to string /mnt/www/application/models/Wae.php

pa8s, 2025-01-30:
**Describe the bug**
While changing worked/confirmed tick boxes, any combination with `Show confirmed` unselected results in Wavelog sends an empty page.

**Version Information**
- OS: FreeBSD 13.4
- PHP: 8.4.2
- MySQL/MariaDB:  11.4.4-MariaDB
- Wavelog: 1.9.1 dev branch, commit 9613e397

**Steps to Reproduce**
1. Go to Awards -> International -> Worked All Europe (WAE)
2. Worked/Confirmed -> Any combination with `Show confirmed` unselected -> Empty page and error logged.

**Expected Behavior**
Able to select any combination of tick boxes

**Log**
`Severity: error --> Exception: Object of class WAE could not be converted to string /mnt/www/application/models/Wae.php 150`

filipmelik, 2025-01-30:
Thanks for detailed report. Was able to reproduce, fix is fairly easy and on the way #1587

HB9HIL, 2025-01-30:
tnx @filipmelik

pa8s, 2025-01-30:
Checked, ok now. Thnx

closed 2025-01-30 by HB9HIL