wavelog/wavelog #1588: WAE Award is missing confirmed IT9 and TA1 qsos

pa8s, 2025-01-30:
**Describe the bug**
While checking WAE Award I noticed IT9 and TA1 qsos are not seen as worked/confirmed

**Version Information**
OS: FreeBSD 13.4
PHP: 8.4.2
MySQL/MariaDB: 11.4.4-MariaDB
Wavelog: 1.9.1 dev branch, commit 13d774db

**Steps to Reproduce**

1. Go to Awards -> International -> Worked All Europe (WAE)
2. Worked/Confirmed -> Select `Show not worked` 
3. Checking listed not worked prefixes, IT9 and TA1 are confirmed in my log

**Expected Behavior**
IT9 and TA1 visible in the `Show confirmed` list and not in `Show not worked`

phl0, 2025-01-30:
Can you please show/check which region is set for your confirmed IT9 / TA1 QSOs?

pa8s, 2025-01-30:
One TA1 QSO : DXCC TA - Turkey - CQ Zone 20 - ITU Zone 39 - Continent Asia - Region <empty>
Other TA1 QSO : DXCC TA - Turkey - CQ Zone 20 - ITU Zone 39 - Continent Europe - Region <empty>

One IT9 QSO : DXCC I - Italy - CQ Zone 15 - ITU Zone 28 - Continent Europe - Region <empty> - IOTA EU-025
Other IT9 QSO : DXCC I - Italy - CQ Zone 15 - ITU Zone 28 - Continent Europe - Region <empty> - IOTA <empty>

phl0, 2025-01-30:
Seems that region is not set correctly for those. Example from my working log:

[image]

pa8s, 2025-01-30:
Is this field filled in automatic at any point? Or is it a manual action? If manual could a function like 'Check QSOs missing continent data' be available for regions (or include this in the continent data function)?

phl0, 2025-01-30:
No at this point it is not filled automatically.

phl0, 2025-01-30:
To be honest: We are still awaiting the publishing of the criteria used to determine region. Without that we cannot code automatism. So at this point you have to fill it manually I am afraid.

phl0, 2025-01-30:
But you could for example use the Logbook Advanced and search for IT9 as DX. Then use edit function to set Region.

pa8s, 2025-01-30:
Hmm, bulk edit in Logbook Advanced did not help. DX IT9 -> select all IT9 qsos -> Edit -> Column Region -> Sicily -> Save
Checked one of those QSOs, Awards -> Region -> empty

phl0, 2025-01-30:
Oh right. The LBA edit seems b0rken here. Will investigate.

pa8s, 2025-01-30:
Ok, so this WAE issue can be closed. Do you create a new issue regarding LBA edit?

closed 2025-01-30 by pa8s

phl0, 2025-01-30:
Bug sound. See https://github.com/wavelog/wavelog/pull/1590. Should make region edit in LBA work.