wavelog/wavelog #1603: Once in a while Gridtracker2 fails to upload FT8 qso to Wavelog

cipman, 2025-02-02:
About 1 in 10-20 FT8 qsos registered by GridTracker v2.250101.1 fails to get uploaded to Wavelog, see picture - https://imgur.com/a/dQjbGv8
A successful upload is recorded by Nginx in access.log file as:
   "POST /index.php/api/qso HTTP/1.1" 201 104 "-" "GridTracker/2.250101.1"
while a failed upload is recorded as:
   "POST /index.php/api/qso HTTP/1.1" 200 52 "-" "GridTracker/2.250101.1"
How can I troubleshoot this?

Version Information:
    OS: linux mint 22.1
    PHP: 8.3.6
    MariaDB: 10.11.8-MariaDB-0ubuntu0.24.04.1
    Wavelog: 10.11.8-MariaDB-0ubuntu0.24.04.1

Steps to Reproduce:
    Do some FT8 qso's (>20) and have Gridtracker2 set to automatically log them to Wavelog (Cloudlog setting).

Expected Behavior:
    I expect all qso's to get logged to Wavelog with no errors.

Screenshots Include screenshots if applicable to illustrate the issue.
    https://imgur.com/a/dQjbGv8

phl0, 2025-02-02:
Please enable logging on your Wavelog instance and see if that reveals some error. Maybe you want to increase lig level as well. See application/config/config.php

cipman, 2025-02-08:
I updated Wavelog from 1.9.1 to 2.0. Same issue.
Here are some log lines I pulled off comparing "OK" and "ERROR" block of lines (OK = logged qso, ERROR = fail to log qso):
https://pastebin.com/4WG4Vksz - from /var/www/html/application/logs/log-2025-02-08.php
https://pastebin.com/YB9vJmK7 - from /var/log/nginx/error.log

vastsea-wuji, 2025-02-12:
I also encounter this problem, and I have to upload the log again to wavelog every day. The logging level of my program is always Debug. I looked at the log and there were no error messages. When I looked at the nginx exception log, I found a large number of nearly identical errors, and these error logs seemed to occur at the same time and in the same number of Qsos that were missed. Here is the error log:
`
2025/02/11 17:09:48 [error] 6819#0: *14242 attempt to set status 403 via ngx.exit after sending out the response status 200, client: 39.144.211.246, server: xxx, request: ＂POST /index.php/api/qso HTTP/1.1＂, host: ＂log.59888888.xyz＂
2025/02/11 18:29:40 [error] 6819#0: *17144 attempt to set status 403 via ngx.exit after sending out the response status 200, client: 39.144.211.246, server: xxx, request: ＂POST /index.php/api/qso HTTP/1.1＂, host: ＂log.59888888.xyz＂
2025/02/12 15:02:18 [error] 2182#0: *10273 attempt to set status 403 via ngx.exit after sending out the response status 200, client: 39.144.210.116, server: xxx, request: ＂POST /index.php/api/qso HTTP/1.1＂, host: ＂log.59888888.xyz＂
2025/02/12 15:18:33 [error] 2182#0: *10816 attempt to set status 403 via ngx.exit after sending out the response status 200, client: 39.144.210.127, server: xxx, request: ＂POST /index.php/api/qso HTTP/1.1＂, host: ＂log.59888888.xyz＂
2025/02/12 15:38:33 [error] 2182#0: *11543 attempt to set status 403 via ngx.exit after sending out the response status 200, client: 39.144.210.127, server: xxx, request: ＂POST /index.php/api/qso HTTP/1.1＂, host: ＂log.59888888.xyz＂
2025/02/12 15:43:03 [error] 2182#0: *11713 attempt to set status 403 via ngx.exit after sending out the response status 200, client: 39.144.210.127, server: xxx, request: ＂POST /index.php/api/qso HTTP/1.1＂, host: ＂log.59888888.xyz＂
2025/02/12 17:39:48 [error] 2182#0: *16083 attempt to set status 403 via ngx.exit after sending out the response status 200, client: 39.144.210.127, server: xxx, request: ＂POST /index.php/api/qso HTTP/1.1＂, host: ＂log.59888888.xyz＂
`
I also had an interesting workaround, which I solved when I restarted gradtracker

cipman, 2025-05-17:
Updated wavelog to v2.0.4, gridtracker2 was at v2.250507.0. 
After 15 qso's in about an hour I got an error, but this time there was  a hint in GT2: "Fail log in Cloudlog // See main.log for error".
In GT2 main.log I found:
   [error] Cloudlog/Wavelog response:
   [error] {"status":"failed","reason":"wrong JSON"}

closed 2025-05-26 by github-actions[bot]