wavelog/wavelog #1605: Advanced Logbook - wrong content in some columns

205er, 2025-02-02:
Hi,

I use the most recent version of Wavelog (2.0) and today I stumbled across the following issue:

I imported some QSO from Ham2K Polo and displayed them in the advanced logbook view, where I noticed, that the content does not match the list header. POTA references are put out as "SOTA" and the column of "POTA" contains only state or country codes. First of all I thought, that the import file was corrupt, but the detail view of the QSO in Wavelog is 100% fine! See screenshots below.

**Screenshots**
[image]
[image]

205er, 2025-02-02:
I use the config functionality to enable/disable some of the column. Do you need to know which of them are enabled/disabled in my Wavelog instance?
I also send out the ADIF file to @HB9HIL by mail

AndreasK79, 2025-02-02:
@205er it would be good if you post which ones are disabled. I can't reproduce this problem here.

205er, 2025-02-02:
[image]
[image]

HB9HIL, 2025-02-02:
>  I also send out the ADIF file to [@HB9HIL](https://github.com/HB9HIL) by mail

I did not rcvd a file yet.. :D

205er, 2025-02-02:
Very Strange: Enabling the "IOTA" field helped to fix this issue:

[image]

After i disabled IOTA again, the values still appear in the correct columns!

[image]

AndreasK79, 2025-02-02:
Caching issues?

205er, 2025-02-02:
Off course, I checked this first! ;-) This issues appeared in three different browsers and even after I cleared the cache.

AndreasK79, 2025-02-02:
We can certainly exclude that then :) Difficult to say, but as long as it works now, not much we can do.

205er, 2025-02-02:
Thanks for your time!!!

closed 2025-02-02 by 205er

AndreasK79, 2025-02-02:
No problem. Report whatever you find. Bugs do exist, we just got to find them :)