wavelog/wavelog #1625: User Setting to Disable "No QSOs Today" Banner.

northrup, 2025-02-07:
**What Feature is Missing?**
The display at the top of the page for "You have made no QSOs today; time to turn on the radio!" is a nice sentiment, however I don't want to see it every time I log into my logging software. This should be a user toggleable feature to enable/disable as the user sees fit, or for an admit to remove the feature all together from the site.

**Why is This Feature Important?**
It adds to a more enjoyable user experience. It could have been that the origin of this feature was as a visual check to alert people if they had uploaded or logged data and it wasn't logged correctly, but there are other ways to surface that data. As a radio operator I know if I made a QSO today operating my radio, I don't need the software to tell me at the top of the page in a banner every time I log into it. 

**Is This Feature Personal or Beneficial to Others as Well?**
I believe this feature would have broad impact and appreciation. I'm not asking for the feature to go away, just to make it user/site configurable.

HB9HIL, 2025-02-07:
This is not missing. You can find the setting in the global options under appearence.

northrup, 2025-02-07:
@HB9HIL 🤔 I saw that, but that entirely disables the notification banner all together for everyone, which would indicate that other notifications beyond "No QSOs Today" would also be suppressed.

I'm not looking to disable all notifications, just the "No QSOs Today" alerting.

HB9HIL, 2025-02-07:
So you want to disable "No QSOs today", but you want to see "You made X QSOs today!"
?

northrup, 2025-02-07:
@HB9HIL No, that's not what I want at all 😂 . I just want the ability to disable the QSO messages without disabling other alerts. Based upon our conversation thus far I came away with "I must clearly be missing something" because it sounded like we were talking past each other, and I found out why. I went looking at the code and discovered (what I'm _positive_ you know) that the QSO messages are the only thing gated behind the "Dashboard Banner" option (https://github.com/wavelog/wavelog/blob/master/application/views/dashboard/index.php#L162-L175). 

It is not clear from the documentation or the in application descriptions that "Dashboard Notification Banner" only applies to the QSO messages and not other messages that also use the alert role. Aside from looking at the code a user would have no way of knowing that. 

I still hold to my position that the QSO message might be useful to some and should probably be a user definable banner option as opposed to "nobody using the installation gets these messages or everyone using the installation gets these messages."

HB9HIL, 2025-02-07:
Okay so the "issue" that probably other dashboard alerts are not affected when disabling the notification banner. I must assume that this could be phrased differently but this is a small thing to change. 

Besides that we currently not planned to make the dashboard notification a user options rather then a global option but we are always open for pull requests 😄

AndreasK79, 2025-05-02:
As this is now implemented as a user setting in the dev branch, this issue will be closed.

closed 2025-05-02 by AndreasK79