wavelog/wavelog #1626: WAE Enties in Wavelog

dl6mhw, 2025-02-07:
**What Feature is Missing?**
WAE Entity Code for a QSO and it derivation from the Calls, statistics fpr WAE and later use the in award application

**Why is This Feature Important?**
The Work All Europa Award is the most important international award of the DARC.  It domanitates also DXCC entites in serveral major contestest. 

**Is This Feature Personal or Beneficial to Others as Well?**
Yes - WAE is useful for many users and also for the reputation of wavelog. The implemantation ist not difficult and was done in the DCL before. I can help id there are questions.

phl0, 2025-02-07:
This is known and we are waiting for a list/definition which is machine-readable and can be imported. We will not make a list manually.

dl6mhw, 2025-02-07:
1. The Definition is in the WAE-Rules: https://www.darc.de/en/der-club/referate/committee-dx/diplome/wae-award/

2. A machine readable list is in the DCL datebase tables, that can be exported. (I do not have access). Ask dcl@darc.de

3. There is trick implemented in DCL
--> the ADIF-Nummer is not defined for WAE-Countries
- therefore the ADIF Number for a WAE-Station e.g. MM/DL6MHW on Shetland-Island is ADIF(Maincountry)+1000. For MM/DL6MHW :1280.
- so you can derive DXCC-Adif by the module 1000 operation
(It works for more then 10 Years)

4. The derivation of WAE
- In some cases it is easy: e.g. IT9 or TA1
- However there are Exception
 -  also IH9 is 'new' for IT9 (Sicily)
  - 4U1A is 4U1V
  - MM/DL6MHW is GM on 1.8.2007 but GM/s on 12.8.2007
- for that there is a database table with WAE-Exception which was managed by the WAE-Award-Manager DF8AA, he defies triples (call, WAE,time-periode) for difficult cases
  - the concept of exceptions is also in clublog - we discussed this topic in 2014 with G7VJR

- for the derivation of WAE-Contry Code there is a function in DCL that could be adopted

5. There are unresolvable problems like African Italy (IG9)

6. For WAE one can also use cty-Files which contains many exceptions for specific WAE-Contries. We do it for Contests like WAE-Contest and WAG.

(Please exuse my poor English - it is difficult for me)

int2001, 2025-02-08:
Re 1) This is well known to us.

Re 2) There are already discussions ongoing to make those lists public. As long as they aren't provided in a structured and machinereadable way, there's no progress here.

Re 3) No way. ADIF-standards says: WAE-Regions have to be stored within the field called region. And that's the way we're going implement a possible derivation. The current built-in awardpage already takes care of that region-field. The user can set the region currently manual. It's all about the derivation. Furthermore: We're not going to use the well-documented field for the DXCC for doing some weird undocumented modulo-calculations. It's all defined within the ADIF-standard. PS: Very creative approach at DCL you did ;)

Re 4) tbh: makes no sense that some single-person has the details about it. An Award should be transparent to everyone. Means at the end the same as point 1: If those exceptions are published in a structured and machinereadable way, we're going to use them.

Re 5) Unsolveable problems cannot be solved with any IT-solution ;)

Re 6) As official known, DARC is the master of WAE-definition, and not Clublog-CTY-Files.

AndreasK79, 2025-05-27:
Since all of this is up to DARC, and they have not provided us with any information, I'll close this issue. If they provide us with information, we will add the necessary enhancements at that time.