wavelog/wavelog #1654: Clublog download does not work

oliverbross, 2025-02-13:
Latest version of wave log as of 13/02/2025

Clublog download via cron or manual does not work, will not download any qso's:

[image]

int2001, 2025-02-13:
Clublog blocks your IP, in case of:
- a few times sync with wrong credentials/
- Trying to upload QSOs which were already uploaded (happens fast, if you forgot to checkmark "already uploaded to clublog")
- Trying to upload QSOs for a call which isn't registered at clublog
- Trying to upload QSOs which are "non-sense" (like wrong DXCC and so on)
- some more unspecified things

we tried to catch the most of those exceptions. but it's currently impossible to catch them all.

the block/ban is normally dropped within a few hours from clublog.
i'd suggest you double-check everything and try it with manual upload.

int2001, 2025-02-13:
This is strange.
any special modes like SAT-QSO or DMR-"Internet-Radio"?
and: Does clublog-interface show confirmation as well?

Or maybe a QSO which was confirmed via clublog "in the future"?
Wavelog only downloads the QSL's from Clublog which were created AFTER your last Clublog QSL within Wavelog.
e.g.: You have 40 unconfirmed QSOs during January. You (accidently) marked a QSO as "Clublog-Confirmed" today --> Wavelog will ask Clublog ONLY for QSLs which were received after today.

Worarkound: Go to LBA (Logbook-Advanced) and remove the most recent QSLs for Clublog there. Start a new sync

int2001, 2025-02-13:
indeed. misunderstanding.

Wavelog doesn't import QSOs from 3rd-party services. it only downloads the confirmations from there. you need to have the QSO in your log.

Wavelog is meant as main-logging tool (like other mainlogs).

closed 2025-02-13 by int2001