wavelog/wavelog #1656: Filtering of spots not working - using enhanced syntax

oliverbross, 2025-02-13:
Wavelog: latest version as of 13/02/2025

The documentation states:

You can easily filter the Spots by typing in search-terms a the searchfield on top of the cluster. There's even an enhanced syntax available. E.g.: Type in !POTA and you won't see POTA Spots. You can also combine terms like !FT8 Italy -> Shows all Spots with DXCC Italy, but no FT8-Spots and so on.

Example:

[image]

now let's try to use the enhanced syntax, in this case !FT8, the results are as follows:

[image]

normal filtering with "!"works, but using "!" shows just empty cluster

int2001, 2025-02-13:
Feature is in dev-branch - dev may contain bugs. That's why it is called dev. It isn't in master yet and there was no release today.

According to your screenshots you are NOT on latest dev.

The enhanced syntax was already tested on several setups. We tested it again as action on your issue here. It is working.

The filtering has only effect on the shown items in cluster. The cluster does not contain any modes (no cluster does that. Mode can(!) only be derived)

We are going to mark the entry at the wiki as "comes with next version".

Thank you for the heads up.

oliverbross, 2025-02-13:
thanks for prompt reply, it would be great to mark it "available in the next version" to avoid any confusion.

closed 2025-02-13 by int2001